MCUdude/MegaCore #120: How to compile bootlaoders?

harryberlin, 2019-09-09:
Hello
I did a mod for `optiboot_flash.c`
After this, i tried to compile on debian stretch with this command
`make -e TOOLROOT=/home/pi/arduino-nightly/hardware/tools/avr`
But the script fails:

>Build 1 of 6318 
Target 1 of 78 
Sub build 0 of 81 
 make atmega8 AVR_FREQ=24000000L BAUD_RATE=1000000 LED=B5 LED_START_FLASHES=2 UART=0 
Using avr-gcc 7.3.0

>UART = UART0	LED_PIN = B5	 LED_FLASHES = 2	 SUPPORT_EEPROM = 	 COPY_FLASH_PAGES = 
\x1B[1m\x1B[4m
BAUD RATE CHECK: Desired: 1000000 Real: 1000000 UBRRL = 2 Difference: 0.0 %\x1B[0m
/home/pi/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: address 0x200a of bootloaders/atmega8/24000000L/optiboot_flash_atmega8_UART0_1000000_24000000L.elf section `.text' is not within region `text'
/home/pi/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: address 0x200a of bootloaders/atmega8/24000000L/optiboot_flash_atmega8_UART0_1000000_24000000L.elf section `.text' is not within region `text'
/home/pi/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: section .version loaded at [00001ffe,00001fff] overlaps section .text loaded at [00001e00,00002009]
collect2: error: ld returned 1 exit status

>/home/pi/arduino-nightly/hardware/tools/avr/bin/avr-size: 'bootloaders/atmega8/24000000L/optiboot_flash_atmega8_UART0_1000000_24000000L.elf': No such file
/home/pi/arduino-nightly/hardware/tools/avr/bin/avr-objcopy: 'bootloaders/atmega8/24000000L/optiboot_flash_atmega8_UART0_1000000_24000000L.elf': No such file
make[1]: *** [bootloaders/atmega8/24000000L/optiboot_flash_atmega8_UART0_1000000_24000000L.hex] Fehler 1
Makefile:1235: die Regel für Ziel „bootloaders/atmega8/24000000L/optiboot_flash_atmega8_UART0_1000000_24000000L.hex“ scheiterte

>COMPILATION FAILED! 
`

jandrassy, 2019-09-09:
your bootloader doesn't fit into memory limit
"section .text' is not within region text"

MCUdude, 2019-09-09:
You can increase the size by adding SUPPORT_ EEPROM=1 to the makefile command

jandrassy, 2019-09-09:
> You can increase the size by adding SUPPORT_ EEPROM=1 to the makefile command

or BIGBOOT=1 ?

harryberlin, 2019-09-09:
got it working wirth this command:
`make atmega2560 AVR_FREQ=16000000L BAUD_RATE=115200 LED=B7 LED_START_FLASHES=2 UART=0 SUPPORT_EEPROM=1`

closed 2019-09-10 by MCUdude