MCUdude/MegaCore #127: PlatformIO

Martinizhr, 2020-03-03:
Hi
I'm using VS Code and PlatformIO to write some code. 
Can someone please help me set it up. 
I tried google-ing for someone with the same problem but can't find anyone. 

As far as I know you are just supposed to edit platform.ini fine.
```
; ENVIRONMENT SETTINGS
;[env:MegaCore]
[env:ATmega2560]
;[env:megaatmega2560]
; platform = atmelavr
platform = https://github.com/platformio/platform-atmelavr.git#develop
; #develop
framework = arduino

; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
; board = megaatmega2560
board_build.mcu = atmega2560
board = ATmega2560
; Clock frequency in [Hz]
board_build.f_cpu = 8000000L
```
As you can see I tried many different combinations but it alwais gives me this error:
`\.platformio\packages\framework-arduino-avr-megacore\cores\MegaCore/Arduino.h:316:26: fatal error: pins_arduino.h: No such file or directory`

Please HELP me, I'm losing my mind here..

MCUdude, 2020-03-03:
Hmm, that's strange. This platformio.ini config works for me!

```ini
[env:MegaCore]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = ATmega2560
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L

; HARDWARE SETTINGS
; Oscillator option
board_hardware.oscillator = external
; Hardware UART for serial upload
board_hardware.uart = uart0
; Brown-out detection
board_hardware.bod = 2.7v
; EEPROM retain
board_hardware.eesave = yes

; UPLOAD SETTINGS
board_upload.speed = 115200
; Upload serial port is automatically detected by default. Override by uncommenting the line below
;upload_port = /dev/cu.usbserial*

; BUILD OPTIONS
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
build_flags = 

; Upload using programmer and bootloader tool
;upload_protocol = usbasp
; Aditional upload flags
;upload_flags = -Pusb

; SERIAL MONITOR OPTIONS
; Monitor and upload port is the same by default
;monitor_port = 
; Serial monitor baud rate
monitor_speed = 9600
```

```

Processing MegaCore (platform: atmelavr; framework: arduino; board: ATmega2560)
---------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega2560.html
PLATFORM: Atmel AVR 2.0.0 > ATmega2560
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 255KB Flash
PACKAGES: 
 - framework-arduino-avr-megacore 2.0.2 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/MegaCore/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.1% (used 9 bytes from 8192 bytes)
Flash: [          ]   0.3% (used 656 bytes from 261120 bytes)
```

Martinizhr, 2020-03-04:
Thanks for the fast response.
It works now.

closed 2020-03-04 by MCUdude