MCUdude/MegaCore #131: ATmega64 Sketch Upload Problem with FTDI (Related with Default Baudrate)

ustaufuk1, 2020-05-17:
Hello, first of all I want to say huge thank you to MCUdude for awesome package.

I am using ATmega64 with MegaCore and there is no problem with burning bootlader. I am doing with Arduino Mega as ISP device and after burning bootlader LED is blinking as excepted. Also no problem to upload sketch without bootlader at ISP mode (with holding shift key). 

The problem starts when I switch to USB-Serial converter which is FTDI FT232R. Problem occurs when I am uploading sketch to the ATmega64. I am using MCU with 16Mhz external oscillator and as you know default baudrate is **115200**. I can not upload the sketch at this baudrate and I realize when I decrease the default baudrate to the **9600** it works like a charm. Also, it is sometimes working at **19200** and sometimes not but at **9600** it is working all the time. I can say converter is working properlly because I am using it with many different devices at **115200**.

No problem at Reset/DTR/Rx/Tx pins. Only solution is decreasing the baudrate.

What could be the problem? Do you have any idea to check somewhere?

I am handle that problem with changing default board settings from `\MegaCore\hardware\avr\2.0.3\boards.txt`

`64.menu.clock.16MHz_external.upload.speed=115200 => 64.menu.clock.16MHz_external.upload.speed=9600`

**------ Error Output ------**

>          avrdude: Version 6.3-20190619
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is C:\Users\***\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.3/avrdude.conf
> 
>          Using Port                    : COM6
>          Using Programmer              : arduino
>          Overriding Baud Rate          : 115200
>          AVR Part                      : ATmega64
>          Chip Erase delay              : 9000 us
>          PAGEL                         : PD7
>          BS2                           : PA0
>          RESET disposition             : dedicated
>          RETRY pulse                   : SCK
>          serial program mode           : yes
>          parallel program mode         : yes
>          Timeout                       : 200
>          StabDelay                     : 100
>          CmdexeDelay                   : 25
>          SyncLoops                     : 32
>          ByteDelay                     : 0
>          PollIndex                     : 3
>          PollValue                     : 0x53
>          Memory Detail                 :
> 
>                                   Block Poll               Page                       Polled
>            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>            eeprom         4    20    64    0 no       2048    8      0  9000  9000 0xff 0xff
>            flash         33     6   128    0 yes     65536  256    256  4500  4500 0xff 0xff
>            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
>            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
> 
>          Programmer Type : Arduino
>          Description     : Arduino
>          avrdude: stk500_recv(): programmer is not responding
>          Hardware Version: 3
>          Firmware Version: 8.0
>          avrdude: stk500_recv(): programmer is not responding
>          Vtarget         : 420030.5 V
>          Varef           : 0.3 V
>          Oscillator      : 28.800 kHz
>          SCK period      : 3.3 us
> 
>          avrdude: stk500_recv(): programmer is not responding
>          avrdude: initialization failed, rc=-1
>          Double check connections and try again, or use -F to override
>          this check.
> 
>          avrdude: stk500_recv(): programmer is not responding
> 
>          avrdude done.  Thank you.
> 
>          Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
> 

**------ Successfull Output ------**

>          avrdude: Version 6.3-20190619
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "C:\Users\***\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.3/avrdude.conf"
> 
>          Using Port                    : COM6
>          Using Programmer              : arduino
>          Overriding Baud Rate          : 9600
>          AVR Part                      : ATmega64
>          Chip Erase delay              : 9000 us
>          PAGEL                         : PD7
>          BS2                           : PA0
>          RESET disposition             : dedicated
>          RETRY pulse                   : SCK
>          serial program mode           : yes
>          parallel program mode         : yes
>          Timeout                       : 200
>          StabDelay                     : 100
>          CmdexeDelay                   : 25
>          SyncLoops                     : 32
>          ByteDelay                     : 0
>          PollIndex                     : 3
>          PollValue                     : 0x53
>          Memory Detail                 :
> 
>                                   Block Poll               Page                       Polled
>            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>            eeprom         4    20    64    0 no       2048    8      0  9000  9000 0xff 0xff
>            flash         33     6   128    0 yes     65536  256    256  4500  4500 0xff 0xff
>            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
>            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
> 
>          Programmer Type : Arduino
>          Description     : Arduino
>          Hardware Version: 3
>          Firmware Version: 8.0
>          Vtarget         : 0.3 V
>          Varef           : 0.3 V
>          Oscillator      : 28.800 kHz
>          SCK period      : 3.3 us
> 
>          avrdude: AVR device initialized and ready to accept instructions
> 
>          Reading | ################################################## | 100% 0.02s
> 
>          avrdude: Device signature = 0x1e9602 (probably m64)
>          avrdude: reading input file "C:\Users\***\AppData\Local\Temp\arduino_build_557971/ATMEGA64_First_Test.ino.hex"
>          avrdude: writing flash (1294 bytes):
> 
>          Writing | ################################################## | 100% 1.82s
> 
>          avrdude: 1294 bytes of flash written
> 
>          avrdude done.  Thank you.
> 

[image]

AnatiliyKsl, 2020-05-19:
The problem may be in FTDI FT232R. Very often there are fakes of this microcircuit.First you need to check the operation of the FTDI FT232R itself. In my developments, I stopped using this chip, due to fakes. The chip identifier was erased very often, and it was not detected.I had to restore it through a special program. Rarely, but it was not matching data rates.When using CH340G, there were no such problems.In all new developments, I began to use only CH340G

ustaufuk1, 2020-05-19:
Thanks for the tip but I am using this FTDI almost 2 years. So, I didn't have any problem until today and after that too. I don't think it is beacuse of that :/

tomaskovacik, 2020-05-20:
so you have bootloader for 9600, program bootloader again, do not forget to change boards.txt back to original (64.menu.clock.16MHz_external.upload.speed=115200) so you have correct bootloader binary selected

MCUdude, 2020-05-26:
@ustaufuk1 are you experiencing the same issues with the previous version of MegaCore (v2.0.2)? There is a known issue with some 1 MHz bootloaders that I'm about to fix, so it would be interesting to know if 2.0.2 works for you or not

ntsang-lab, 2020-08-14:
thanks for helping me with the Burn Bootloader problem.
Currently I cannot upload code from arduino Ide to my atmega 64
 I connected my arduino and arduino 64 as follows.
 5v-5v
GND-GND 
Tx-Rx
Rx-Tx
Reset- Reset  

I installed TOOL as shown below and press CTRL + U
[image]

per1234, 2020-08-14:
>Tx-Rx
Rx-Tx

For this specific application, where you are using the Uno as a USB to serial adapter, this wiring is actually wrong. You need to make the connections TX-TX, RX-RX.

It's confusing because we always make RX-TX, TX-RX connections, but in this case you are communicating between the ATmega16U2 chip on your Uno and the ATmega64. The pin labels on your Uno are for the ATmega328P. Since the ATmega16U2 is connected to the ATmega328P with RX-TX, TX-RX, the pin marked "RX" on your Uno is actually connected to the TX pin of the ATmega16U2, and the same with the pin marked "TX". So when you make that RX-RX, TX-TX connection between the Uno and the ATmega64, according to the labels on the Uno's silkscreen, you are actually making an RX-TX, TX-RX connection between the ATmega16U2 and the ATmega64.

>Reset- Reset

As for the reset-reset connection, I guess that will work for the auto-reset, I'm not sure because I own dedicated USB to serial adaptersso I never need to do this hacky thing of using an Arduino board for that purpose. It's common for people to connect the reset pin to ground on the Uno when they are using it in this type of application, to hold the ATmega328P in reset so it can't interfere with communication between the ATmega16U2 and the ATmega64, but that would also interfere with the auto-reset capability. So if you're going to do this you should at least upload a **File > New** sketch to your Uno to make sure the sketch isn't doing any serial communication.

pratikgharte, 2021-07-28:
Successfully uploaded the code to ATmega64 but Code is not executing it's functionality.

MCUdude, 2022-08-19:
OP didn't respond, so I'll assume the issue has been resolved. Either way, it was probably a hardware problem.

closed 2022-08-19 by MCUdude