MCUdude/MegaCore #137: Add new "AVR-DA-core for avr128DA family

wa2mze, 2020-07-21:
You can add GCC support by following the method shown here:
https://gist.github.com/goncalor/51e1c8038cc058b4379552477255b4e1

This guide is for the atmega328pb, but I used a similar method to add support for the avr128DA on my computer.  You can download the required .o, .h, and .a files from Microchip and copy them into the avr-gcc package.  Then edit io.h to add the required elif cases for the new chips.  This will grab the required files from Microchip's own mod's to avr-gcc and avr-libc.

You will also have to mod the Arduino core files for the new chip, and others will have to mod their lib's.  But it looks do-able.

MCUdude, 2020-07-21:
Well, the DA and DB series isn't really the scope of this core. However, the latest Arduino toolchain supports the DA series at least. Not sure about the DB's yet. Just update the Arduino AVR package in the boards manager, and you're ready!

I also know @SpenceKonde is working on an Arduino core for the DA/DB series. Not sure when it will be public though.

closed 2020-07-21 by MCUdude

SpenceKonde, 2020-07-21:
DxCore will be released by end of this weekend.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

MCUdude, 2020-07-26:
> DxCore will be released by end of this weekend.

Patiently waiting!