MCUdude/MegaCore #138: Please help me

ntsang-lab, 2020-08-13:
I use atmega 64 .I have a little question in this schematic photo
https://camo.githubusercontent.com/9407f77bb6316f8ea42b45d72acb6212b8baf726/68747470733a2f2f692e696d6775722e636f6d2f426b4a664957432e706e67

Please explain me the tag "AVR ISP header"Why are you connecting differently with the connection pictures of atemega 2560 and minicore?
https://camo.githubusercontent.com/a950c6f8840292f99c5b57a3ea354a439eac5c4b/687474703a2f2f692e696d6775722e636f6d2f675153314f52762e706e67

I have connected my arduino uno to atmega 64 as follows:
 GND-GND 
5v-5v 
D10 -RESET 
D11-MOSI
D12 -MISO
D13 -SCK 
I have checked and made sure there is no wrong connection
 After burning the bootloader, it failed
I want to use arduino UNO to burn bootloader for atmega 64. Please guide me.
Thank you

ntsang-lab, 2020-08-13:
How do I contact you. Give me your Facebook or email! 
my email :ntsangbg98@gmail.com

MCUdude, 2020-08-13:
> How do I contact you. Give me your Facebook or email!
my email :ntsangbg98@gmail.com

That's not how it works. Any help/support happens here.

I can't explain _why_ the ATmega64 (and all the 64 pin parts) don't use MOSI and MISO for data, but rather PDI and PDO. All I know is that's a fact. I know this because I've designed boards for an ATmega128 that didn't work because of this

From the datasheet:

> • PDO/TXD0 – Port E, Bit 1
PDO, SPI Serial Programming Data output. During Serial Program Downloading, this pin is used as data output line for the ATmega64.
TXD0, UART0 Transmit Pin.

• PDI/RXD0 – Port E, Bit 0
PDI, SPI Serial Programming Data input. During serial program downloading, this pin is used as data input line for the ATmega64.

Follow the minimal setup schematic like I've drawn it (using PDI and PDO), and it will probably work.

ntsang-lab, 2020-08-14:
Thank you for your time helping me!!!!!!!!
I am trying to follow your instructions but have not been successful...
I use arduino UNO R3. I connected my arduino and your minimal setup
schematic as follows:
5v-5V
GND-GND
MOSI - PDI  (going through 1k resistor (I tried R = 0))
MISO -PD0
SCK - PB1 (SCK)
RST - RESET

I have tried, but with no success
After loading the example file "ArduinoISP" into my arduino..
 I install programmer: "Arduino as ISP (Mega Core)" then press "burn
bootloader"
all of the above steps are unsuccessful .I checked many times

MCUdude, 2020-08-14:
Is the ATmega64 brand new, or have you desoldered it from an old board? By default, the ATmega64 is running at its internal 1 MHz clock. However, if it's set up to use an external crystal or clock, and no such signal is applied, the microcontroller won't run at all. You can connect a clock signal to the XTAL1 pin on the ATmega64, preferrably above 1 MHz.

if you use [this Arduino as ISP code](https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/ArduinoISP), it's modified to output an 8 MHz clock signal on digital pin 9 on the Arduino UNO. Connect digital pin 9 on the UNO to the XTAL1 pin on the ATmega64 and try again

ntsang-lab, 2020-08-14:
I use ATmega64 brand new 
I design the board to match your minimal scheme
do you need to check the error report on my Arduino IDE

jandrassy, 2020-08-14:
wire pin selected as reset pin in ArduinoISP to reset pin of target

ntsang-lab, 2020-08-14:
sorry i don't understand what you mean
 my english is bad

jandrassy, 2020-08-14:
> 
> 
> sorry i don't understand what you mean
> my english is bad

10- ->RST

ntsang-lab, 2020-08-14:
my arduino -atmega64A
5v-5V
GND-GND
D11 - PDI (going through 1k resistor (I tried R = 0))
D12 -PD0
D13 - PB1 (SCK)
D10 - RESET

ntsang-lab, 2020-08-14:
Arduino: 1.8.13 (Windows 10), Board: "ATmega64, Yes (UART0), BOD 2.7V, LTO disabled, External 16 MHz"

avrdude: Version 6.3-20190619

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\NTS\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.2/avrdude.conf"

         Using Port                    : usb

         Using Programmer              : stk500v2

avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

Error while burning bootloader.

tomaskovacik, 2020-08-14:
few comments:

**as you using Arduino as isp, do not forget 10uF capacitor on reset pin of Arduino which you used as programer (this is to avoid resetting this Arduino, and it is essential!)**

from my little experience boot loading new chips, I always have program related to programming speed ( CLK speed on ICSP)

In Adafruit sketch which MCUdude mentioned there is an option to do slow programming:

https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/26486af1e309d2660efd432a11dc871796939d84/ArduinoISP/ArduinoISP/ArduinoISP.ino#L42 

uncoment line 65: #define LOW_SPEED 

here is how I boot-load new chips:

I use USBasp

select 1MHZ internal oscillator for your target chip 
select USBasp slow (this one can be the problem for you as u use Arduino as isp)
burn bootloader

select 16MHZ external oscillator
select USBasp slow
burn bootloader

after this step normal speed for uploading works

per1234, 2020-08-14:
>avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

This indicates you have **Tools > Programmer > AVRISP mkII (MegaCore)** selected. However, your replies indicate you are using an "Arduino as ISP" programmer. So you need to select **Tools > Programmer > Arduino as ISP (MegaCore)** instead.

>select USBasp slow (this one can be the problem for you as u use Arduino as isp)

Please only do that if you are using a USBasp programmer, which I can see you are not. So ignore that.

tomaskovacik, 2020-08-14:
@per1234 yep, you are right i mix it up :) I will edit my post to fix this

ntsang-lab, 2020-08-14:
Thank you everyone for helping me !
The problem was solved first I burn the bootloader with 1MHz frequency, after successfully I burn the external 16MHz bootloader. Thanks for everyone's help

MCUdude, 2020-08-14:
Thanks, @per1234 and @tomaskovacik for helping out while I was out trying to get my 1yo daughter to sleep!

closed 2020-08-14 by MCUdude

ntsang-lab, 2020-08-14:
thanks for helping me with the Burn Bootloader problem.
Currently I cannot upload code from arduino Ide to my atmega 64
I connected my arduino and arduino 64 as follows.
5v-5v
GND-GND
Tx-Rx
Rx-Tx
Reset- Reset

I installed TOOL as shown below and press CTRL + U
[image]

ntsang-lab, 2020-08-14:
thanks for helping me with the Burn Bootloader problem.
Currently I cannot upload code from arduino Ide to my atmega 64
I connected my arduino and arduino 64 as follows.
5v-5v
GND-GND
Tx-Rx
Rx-Tx
Reset- Reset

I installed TOOL as shown below and press CTRL + U

per1234, 2020-08-14:
Now that I see you also commented the same thing, I'll move my comment over here where it's more appropriate than the other issue you cross-posted to (https://github.com/MCUdude/MegaCore/issues/131):

>Tx-Rx
Rx-Tx

For this specific application, where you are using the Uno as a USB to serial adapter, this wiring is actually wrong. You need to make the connections TX-TX, RX-RX.

It's confusing because we always make RX-TX, TX-RX connections, but in this case you are communicating between the ATmega16U2 chip on your Uno and the ATmega64. The pin labels on your Uno are for the ATmega328P. Since the ATmega16U2 is connected to the ATmega328P with RX-TX, TX-RX, the pin marked "RX" on your Uno is actually connected to the TX pin of the ATmega16U2, and the same with the pin marked "TX". So when you make that RX-RX, TX-TX connection between the Uno and the ATmega64, according to the labels on the Uno's silkscreen, you are actually making an RX-TX, TX-RX connection between the ATmega16U2 and the ATmega64.

>Reset- Reset

As for the reset-reset connection, I guess that will work for the auto-reset, I'm not sure because I own dedicated USB to serial adaptersso I never need to do this hacky thing of using an Arduino board for that purpose. It's common for people to connect the reset pin to ground on the Uno when they are using it in this type of application, to hold the ATmega328P in reset so it can't interfere with communication between the ATmega16U2 and the ATmega64, but that would also interfere with the auto-reset capability. So if you're going to do this you should at least upload a **File > New** sketch to your Uno to make sure the sketch isn't doing any serial communication.

tomaskovacik, 2020-08-14:
@ntsang-lab

a selected programmer is irrelevant for CTRL+U (CTRL+U = upload using serial bootloader, so via selected serial port)

best is to use separate USB to serial converter

what you try to do will only work if you remove ATmega328(168) from your Arduino, so only for DIP version (and I am not sure about RX and TX connection, if it will not work try swapping them)

if you are not able to do that (SMD version for example) you must not connect reset pins together. Instead, pull reset pin of Arduino to the ground and then you have to manually reset atmeg64 shortly before upload (this will probably require few tries before you get timing right :), I personally hit reset after I hit upload in Arduino IDE). You can also power cycle atmega64, it is same as reset.

You can try this program but I did not test it, so there is a chance it will not work

```
/*
 * 10 RX
 * 11 TX
 * 12 RESET
 */

#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX

void setup() {
  pinMode(12, OUTPUT);
  digitalWrite(12, LOW);
  Serial.begin(115200);
  Serial1.begin(115200);
  digitalWrite(12, HIGH);
}

void loop() {
  if (Serial.available()) {      // If anything comes in Serial (USB),
    mySerial.write(Serial.read());   // read it and send it out Serial1 (pins 0 & 1)
  }

  if (mySerial.available()) {     // If anything comes in Serial1 (pins 0 & 1)
    Serial.write(mySerial.read());   // read it and send it out Serial (USB)
  }
}
```

@per1234: empty sketch will not help, the bootloader will always run after reset and screw up communication

ntsang-lab, 2020-08-14:
I have an arduino DIP (I removed the atmega 328p from the board) and connected it like your instructions but it doesn't work.
 I want to ask if it is necessary to change the "programer" in the TOOL menu. I am going to be "AVR ISP".

ntsang-lab, 2020-08-14:
Currently I have bought 1 usb uart using CH340G chip (as picture) does anyone have uploaded the code to atemga 64 by usb ch340g please help me
[image]

ntsang-lab, 2020-08-14:
Arduino: 1.8.12 (Windows 10), Board: "ATmega64, Yes (UART0), BOD 2.7V, LTO disabled, External 16 MHz"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\NTS\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\NTS\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\NTS\Documents\Arduino\libraries -fqbn=MegaCore:avr:64:bootloader=uart0,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\NTS\AppData\Local\Temp\arduino_build_441455 -warnings=none -build-cache C:\Users\NTS\AppData\Local\Temp\arduino_cache_232442 -prefs=build.warn_data_percentage=75 -verbose C:\Users\NTS\AppData\Local\Temp\arduino_modified_sketch_272187\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\NTS\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\NTS\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\NTS\Documents\Arduino\libraries -fqbn=MegaCore:avr:64:bootloader=uart0,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\NTS\AppData\Local\Temp\arduino_build_441455 -warnings=none -build-cache C:\Users\NTS\AppData\Local\Temp\arduino_cache_232442 -prefs=build.warn_data_percentage=75 -verbose C:\Users\NTS\AppData\Local\Temp\arduino_modified_sketch_272187\Blink.ino
Using board '64' from platform in folder: C:\Users\NTS\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.2
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\NTS\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.2
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega64 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega64 -DARDUINO_ARCH_AVR "-IC:\\Users\\NTS\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\NTS\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.2\\variants\\64-pin-avr" "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega64 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega64 -DARDUINO_ARCH_AVR "-IC:\\Users\\NTS\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\NTS\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.2\\variants\\64-pin-avr" "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455\\sketch\\Blink.ino.cpp" -o "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega64 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega64 -DARDUINO_ARCH_AVR "-IC:\\Users\\NTS\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\NTS\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.2\\variants\\64-pin-avr" "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455\\sketch\\Blink.ino.cpp" -o "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455\\sketch\\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core: C:\Users\NTS\AppData\Local\Temp\arduino_cache_232442\core\core_e638d3af682a87837de877948115b39b.a
Linking everything together...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -w -Os -Wl,--gc-sections -mmcu=atmega64 -o "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455/Blink.ino.elf" "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455\\sketch\\Blink.ino.cpp.o" "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455/..\\arduino_cache_232442\\core\\core_e638d3af682a87837de877948115b39b.a" "-LC:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455" -lm
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455/Blink.ino.elf" "C:\\Users\\NTS\\AppData\\Local\\Temp\\arduino_build_441455/Blink.ino.eep"
[... cut ...]
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xd9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xd9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xd9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xd9
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

tomaskovacik, 2020-08-14:
if you removed atmega from uno, then do not forget reset to reset of atmega64, if it do not works, try swaping rx and txt pins

ntsang-lab, 2020-08-14:
> if you removed atmega328 from uno, then do not forget reset to reset of atmega64, if it does not work, try swapping rx and txt lines

I have tried changing Tx and Rx but with no success.
 let me ask when I press CTRL + SHIFT + U does it delete the bootloader of atmega 64 or not

tomaskovacik, 2020-08-14:
CTRL + SHIFT + u is not working in my arduino (1.8.12)
but
if I hit caps lock and so it is ctrl+shift+U (capital U), then it try to upload sketch using programer, which will overwrite bootloader, 

t.