MCUdude/MegaCore #140: From Mega2560 to Mega128 with Megacore = millis() problem

Bern2, 2020-09-05:
Hi,

I have some code running on a Mega2560 pro board that I wanted to port to a Mega128 using Megacore to further integrate it in a HW design I have. Everything went well except 2 issues:

1/ I'm using Timer1 in a statemachine with also use of timer1-OVF for a 'timeout' and reset. Changing TIMSK1 = bit (TOIE1) to TIMSK = bit (TOIE1) solved the issue and that part of the code runs as expected now on the M128.

2/ I'm using millis() throughout another part of the code, but when I set TOIE1 (for timer1) => I can't use millis() anymore (which is supposed to use timer0) and I don't have a serial monitor to further understand what is happening on the M128...

The code below illustrates the issue => configure for the Mega2560 pro board and the led blinks, do it for the M128 and it stops :-/.

`
//const int ledPin =  44;               // Led on the M128 board
const int ledPin =  13;                 // Led on the mega2560  arduino board
int ledState = LOW;                     // ledState used to set the LED
unsigned long previousMillis = 0;  // last time LED was updated
const long interval = 500;           // interval at which to blink (milliseconds)

void setup()
{ pinMode(ledPin, OUTPUT);         // set the digital pin as output:

  ////Timer1 setup: module configuration
  TCCR1A = 0;
  TCCR1B = 0;                                    // Disable Timer1 module
  TCNT1  = 0;                                     // Set Timer1 preload value to 0 (reset)

//  TIMSK = bit (TOIE1);                      // configuration of TIMSK for mega128 Timer1
  TIMSK1 = bit (TOIE1);                      // configuration of for a mega2560  arduino board
}

ISR(TIMER1_OVF_vect)
{ //Case 0 of state machine in code to reset a decoding process
  TCCR1B = 0;
}  

void loop()
{ unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    if (ledState == LOW) ledState = HIGH;
    else ledState = LOW;
  }
  digitalWrite(ledPin, ledState);
}`

per1234, 2020-09-05:
Also posted here: https://forum.arduino.cc/index.php?topic=703781

tomaskovacik, 2020-09-05:
I am not expert but will not this set  OCIE0 TOIE0 to zero? (so disable OVF for T0 = millis() disabled)

correct should be TIMSK |= _BV(TOIE0); or TIMSK |= (1<<TOIE1);  no?

// TIMSK = bit (TOIE1); // configuration of TIMSK for mega128 Timer1
TIMSK1 = bit (TOIE1); // configuration of for a mega2560 arduino board

Bern2, 2020-09-05:
Well Tomas, thank you very much!  Spend more than half a day to narrow the problem down and thanks to your input, my problem was solved in less than 5 minutes!! Guess I need to review how to deal with registers correctly, but it is funny all worked well in the M2560...

closed 2020-09-05 by Bern2