MCUdude/MegaCore #141: Arduino ide atmega 128 uart troubles with uart0

punxer, 2020-09-06:
https://electronix.ru/forum/index.php?app=forums&module=forums&controller=topic&id=37536
This code with redefined io pins baud 9800, spi and sd card works good, delays work good with 1 and 8 mhz internal rc, with external rc 16 mhz all code doesnt work properly. But i receive bad nmea strings from uart .8n1 settings. I defined fcpu with needed freq. Gps are sending nmea strings every second with 9600 and 8n1, i can read gps strings at pc with this settings . No boot at mega 128, tryed alot of fuses combination at 1 and 8 interbal rc. Thanx in advance

MCUdude, 2020-11-17:
Sorry for the late reply. There is no reason why it shouldn't work with an external oscillator. The fuse settings in Arduino IDE should be sufficient. You should not have to manually define F_CPU. Note that you'll have to select your oscillator option under the Tools menu in Arduino IDE for the code to work properly.

Do you have a frequency counter or an oscilloscope you can use to measure the frequency of the external oscillator? Seems to me like it is either too slow or too fast. It would also be great if you could measure how long delay(1) and delay(1000) actually last to get an impression on how bad timing really is.

MCUdude, 2021-07-21:
I'm closing this issue due to no response from OP.

closed 2021-07-21 by MCUdude