MCUdude/MegaCore #142: MCUdude boards does not compile in Arduino IDE

MojaLuna, 2020-10-20:
Hi,

I am using RadioHead library and and one function is:
(driver.recv(buf, &buflen))

If I remove '&' before buflen so that value is not saved and I choose any of MCUdude boards, there is a compile error in Arduino IDE which says: "invalid conversion from 'uint8_t {aka unsigned char}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]" and the funny thing is that it compiles just fine if I choose any of Arduino boards (Uno, Mega2560,...) with same code and MCs are working correctly.

I just can not go trough this by myself.

Thank you!

MCUdude, 2020-10-20:
None of my Arduino cores uses the -fpermissive flag. You can read why here:
https://github.com/MCUdude/MiniCore/issues/97#issuecomment-678582518

There is a problem with your code, and MaegaCore doesn't convert this into a warning.

```
"invalid conversion from 'uint8_t {aka unsigned char}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]"
```

Where did you get your library from?

MojaLuna, 2020-10-20:
I got the library from official site https://www.airspayce.com/mikem/arduino/RadioHead/ and library is downloaded from there http://www.airspayce.com/mikem/arduino/RadioHead/RadioHead-1.113.zip.

Thank you for fast response.

jandrassy, 2020-10-20:
`(driver.recv(buf, &buflen))` is in your sketch?

MojaLuna, 2020-10-20:
>  Za:   MCUdude/MegaCore <MegaCore@noreply.github.com>
>  Kopija:   MojaLuna <bojan@tursic.net>, Author <author@noreply.github.com>
>  Poslano:   20.10.2020 17:15

Hi,

Yes, this is in my sketch but this is also in example sketch for receiver which comes with RadioHead library.

Regards,
Bojan

jandrassy, 2020-10-20:
and you agree that `uint8_t` is a very different type then `uint8_t*`?

MojaLuna, 2020-10-20:
Of course, last one is a pointer.
But still I have no idea what is wrong, sorry.

 Poslano:   20.10.2020 17:51

jandrassy, 2020-10-20:
the function is ` bool recv(uint8_t* buf, uint8_t* len);`
so the second parameter should be a pointer to uint8_t variable.
it is strange to use a pointer to a variable for input of buffer size and output of bytes read , common is to use a reference. even more common is to return count of bytes read as return value (with negative values for error codes)
but all this is off-topic here. 
using buflen instead of &buflen is very wrong and the compiler settings in Arduino AVR core are very benevolent. MCUDude has it right.

MojaLuna, 2020-10-20:
OK, thank you for your time. Much appreciated.

I will take it from here.

Best regards!

 Poslano:   20.10.2020 19:23 
 Zadeva:   **SPAM**  Re: [MCUdude/MegaCore] MCUdude boards does not compile in Arduino IDE (#142)

closed 2020-10-20 by MCUdude