MCUdude/MegaCore #144: [ATmega2560] Trouble switching to MegaCore

BIGStrawberry, 2020-11-19:
Hello,

I'm having some trouble switching to MegaCore. When uploading a test sketch to my board with the normal Arduino core it manages to upload without any issues. When I try this with the MegaCore version, I keep receiving the error `avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03` (The value behind `resp=` isn't constant).

Some setup information, I am: 
- using an ATmega2560 board, without anything connected to it;
- using a USB cable to upload my sketch (board has the ATMEGA16U2 as the USB to Serial converter);
- using the blink example sketch;
- using Windows 10 (not sure if that's relevant)

Some of the things I've tried so far;
- made sure the board is not broken by using the normal arduino core and uploading a sketch without issue;
  - I've tried another atmega2560 board (CH430-version), which also had no issues.
- tried both the arduino IDE and platformIO ([with this platformio.ini config file](https://github.com/MCUdude/MegaCore/issues/127#issuecomment-594084013)). This also results in the same error.

 I'm unsure what steps to take next and hoping someone here could point me in the right direction. Thanks in advance.

_Output from PlatformIO using 'MegaCore > ATmega2560' board with the 'AVRISP mkll (MegaCore)' programmer_
```
> Executing task: C:\Users\name\.platformio\penv\Scripts\pio.exe run --verbose --target upload --environment MegaCore <

Processing MegaCore (platform: atmelavr; framework: arduino; board: ATmega2560; board_build.f_cpu: 16000000L; board_hardware.oscillator: external; board_hardware.uart: uart0; board_hardware.bod: 2.7v; board_hardware.eesave: yes; board_upload.speed: 115200; build_unflags: -flto; build_flags: ; monitor_speed: 9600)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega2560.html
PLATFORM: Atmel AVR (3.0.0) > ATmega2560
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 255KB Flash     
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr-megacore 2.0.4
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio\build\MegaCore\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.1% (used 9 bytes from 8192 bytes)
Flash: [          ]   0.2% (used 600 bytes from 261120 bytes)
.pio\build\MegaCore\firmware.elf  :

section                    size      addr

.data                         0   8389120

.text                       600         0

.bss                          9   8389120

.comment                     48         0

.note.gnu.avr.deviceinfo     64         0

.debug_info                3004         0

.debug_abbrev              2842         0

.debug_line                  26         0

.debug_str                  998         0

Total                      7591
<lambda>(["upload"], [".pio\build\MegaCore\firmware.hex"])
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
BeforeUpload(["upload"], [".pio\build\MegaCore\firmware.hex"])
Auto-detected: COM6
avrdude -v -p atmega2560 -C C:\Users\name\.platformio\packages\tool-avrdude\avrdude.conf -c arduino -b 115200 -D -P "COM6" -U flash:w:.pio\build\MegaCore\firmware.hex:i

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:24:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\name\.platformio\packages\tool-avrdude\avrdude.conf"

         Using Port                    : COM6
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xf1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xf1

avrdude done.  Thank you.

*** [upload] Error 1
```

MCUdude, 2020-11-19:
MegaCore uses a different bootloader than the original Arduino MEGA 2560 does.
MegaCore uses Optiboot while the MEGA2560 ships with stk500v2.

You'll have to burn the bootloader again, so it's compatible. You can do this in PlatformIO as well as Arduino IDE.
The [latest PlatformIO template](https://github.com/MCUdude/MegaCore/blob/master/PlatformIO.md#platformioini-template) makes this really easy for you.

MCUdude, 2020-11-20:
If you want to keep the original bootloader you can change the upload protocol.

Instead of `upload_protocol = arduino`, use `upload_protocol = stk500v2` instead.

BIGStrawberry, 2020-11-24:
Thanks for your swift respone @MCUdude ! My issue has been resolved.

closed 2020-11-24 by BIGStrawberry