MCUdude/MegaCore #145: Export Compiled Binary - 2.0.4

DirtyEngineer, 2020-12-01:
I seem to have found a bug with the latest release (2.0.4) when using the "Export Compiled Binary" in the Arduino IDE (v1.8.12 & v1.8.13 tested Win10). 

Programs will verify without issues but when "Exporting Compiled Binary" I get a "The system cannot find the file specified.
Error compiling." error. I was able to track this down to the IDE looking for the file in the Temp directory to copy it to the program directly. It is looking for in my case "Dispenser_v1.2.ino_atmega128_7372800L.lst" but when I check the temp directory the file names there do not have the "__atmega128_7372800L" appendage that they should have. (Example "Dispenser_v1.2.ino.lst").

Not sure if this is related but I also noticed that in the bottom right corner of the IDE it only show "ATmega128 on Com16" instead of the normal "ATmega128, No bootloader, BOD 2.7, LTO disabled, External 7.3728 Mhz on COM16" that shows with MegaCore 2.0.3.

Reverting back to version 2.0.3 fixes the issue and produces a compiled hex in the program folder when using "Export Compiled Binary"

Also its seems that I have the identical issue with MightyCore 2.0.6.  The previous version 2.0.5 will export a compiled binary without issue.

MCUdude, 2020-12-01:
It's a known issue that affects all my cores except MegaCoreX. I'll see if I can get around and do some new releases to fix this once and for all.

If you can't wait, just do a manual install. The bug was fixed in 2f187a4

DirtyEngineer, 2020-12-01:
Thank you for your quick response. 
I have no issue running the previous release. It was just a bit of a head scratcher when I updated to the latest version and I hoped to save others from trying to figure it out.

@MCUdude Thank you so much for your work with these cores. They have been unimaginably helpful.

MCUdude, 2020-12-02:
The latest release, v2.0.7 is now available through boards manager!

closed 2020-12-02 by MCUdude