MCUdude/MegaCore #146: Problem using 1Wire-libs

Greece2, 2020-12-08:
Hi, I'm not sure if it is the Board lib (MegaCore, MiniCore) or if the problem lies in the libraries to read 1Wire-Sensors.
Using the ATmega128 or the ATmega328 from the MegaCore for example, I get the following error messages (only the first lines copied):

> In file included from W:\libraries\DS18B20\src/DS18B20.h:4:0,
>                  from W:\libraries\DS18B20\src\DS18B20.cpp:1:
> W:\libraries\DS18B20\src\DS18B20.cpp: In member function 'void DS18B20::setResolution(uint8_t)':
> C:\Users\XYZ\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:133:22: error: 'l' was not declared in this scope
>    typeof (low) _l = (l);           \
>                       ^
> W:\libraries\DS18B20\src\DS18B20.cpp:104:18: note: in expansion of macro 'constrain'
>      resolution = constrain(resolution, 9, 12);
>                   ^~~~~~~~~

Using other Board libs, like the "Arduino Mega 2560" or the "ATmega328 on a breadboard" , everything compiles fine. So, could it be the MegaCore / MiniCore lib that's causing the problem ?

Thanks for your help !

MCUdude, 2020-12-08:
A bug in the constrain() and round() macros was introduced in the latest release for all my cores. I hope I can have it fixed in all cores (MightyCore, MiniCore, MegaCore, MajorCore, and MicroCore) by tonight. Meanwhile, you can try the previous boards manager release.

Greece2, 2020-12-09:
Thanks for the superfast reply !  I'll roll back to the previous version for now.

MCUdude, 2020-12-17:
MegaCore v2.0.6 is now available!

closed 2020-12-17 by MCUdude