MCUdude/MegaCore #147: Uploading: File not found

slabbi, 2020-12-09:
vrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "C:\Users\STEPHA~1\AppData\Local\Temp\arduino_build_647158/xxx.ino.with_bootloader.hex"
avrdude: can't open input file C:\Users\STEPHA~1\AppData\Local\Temp\arduino_build_647158/xxx.ino.with_bootloader.hex: No such file or directory
avrdude: read from file 'C:\Users\STEPHA~1\AppData\Local\Temp\arduino_build_647158/xxx.ino.with_bootloader.hex' failed

avrdude done.  Thank you.

Bootloader is set to: No Bootloader

slabbi, 2020-12-09:
It seems to be necessary to delete the "preferences.txt" file. 
Otherwise you might get this error.

slabbi, 2020-12-09:
Ok, after removing "preferences.txt" I can flash the code, but when I choose "export compiled binary" I get this

"Der Sketch verwendet 126496 Bytes (48%) des Programmspeicherplatzes. Das Maximum sind 262144 Bytes.
Globale Variablen verwenden 5645 Bytes (68%) des dynamischen Speichers, 2547 Bytes für lokale Variablen verbleiben. Das Maximum sind 8192 Bytes.
xxx.ino.with_bootloader.hex konnte nicht gefunden werden
        1 Datei(en) kopiert.
"

slabbi, 2020-12-09:
Damn, after closing and starting the IDE again, the first error appears again.

slabbi, 2020-12-09:
Fix:
Revert following lines in platform.txt:

- tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" -v {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i"
+ tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" -v {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"

Remove:

## Save hex
recipe.hooks.savehex.presavehex.1.pattern.windows="{runtime.platform.path}/scripts/delete_merged_output.bat" {build.export_merged_output} "{build.path}\{build.project_name}.with_bootloader.hex"
recipe.hooks.savehex.presavehex.2.pattern.windows=cmd /C copy "{build.path}\{build.project_name}_{build.mcu}_{build.f_cpu}.lst" "{sketch_path}"
recipe.hooks.savehex.presavehex.1.pattern.linux=chmod +x "{runtime.platform.path}/scripts/delete_merged_output.sh"
recipe.hooks.savehex.presavehex.2.pattern.linux="{runtime.platform.path}/scripts/delete_merged_output.sh" {build.export_merged_output} "{build.path}/{build.project_name}.with_bootloader.hex"
recipe.hooks.savehex.presavehex.3.pattern.linux=cp "{build.path}/{build.project_name}_{build.mcu}_{build.f_cpu}.lst" "{sketch_path}"
recipe.hooks.savehex.presavehex.1.pattern.macosx=chmod +x "{runtime.platform.path}/scripts/delete_merged_output.sh"
recipe.hooks.savehex.presavehex.2.pattern.macosx="{runtime.platform.path}/scripts/delete_merged_output.sh" {build.export_merged_output} "{build.path}/{build.project_name}.with_bootloader.hex"
recipe.hooks.savehex.presavehex.3.pattern.macosx=cp "{build.path}/{build.project_name}_{build.mcu}_{build.f_cpu}.lst" "{sketch_path}"

Still testing... But looks good.

MCUdude, 2020-12-09:
I'm pretty sure this is an issue with the IDE itself. I tried to reproduce this myself, and "my" IDE always used the ATtiny13/MicroCore platform.txt file when I uploaded using a programmer. I cleared the cache and removed preferences.txt, and that solved my issue.

Note that if the IDE uses the MegaCore platform.txt file, there will always be a `*with_bootloader.hex` file, even though the user has selected "No Bootloader". The No bootloader option also adds a hex file that can be merged with the compiled hex, but it's empty and only contains a header (or footer, not sure).

MCUdude, 2020-12-09:
Aha!

You'll have to make sure you chose a programmer that's bundled with MegaCore, for instance, USBasp (MegaCore). If not, it will use platform.txt from another package. In my case, I was previously working with MicroCore when I suddenly switched over to MegaCore without changing the programmer.

In the preferences.txt file, the selected programmer was:
```
programmer=MicroCore:usbasp
```

Which pretty much is the root of the problem. In your case, try selecting a programmer in the list that has MegaCore in its name:

[image]

slabbi, 2020-12-09:
Clearing the cache only solved the problem one time until the IDE has been restarted.
There is definitely *not* always a *with_bootloader.hex file.
In my case it creates no *with_bootloader.hex file when "No bootloader" is selected.
Changing 
- tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" -v {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i"
+ tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" -v {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"
solved this issue (it is the same as in 2.0.2), which I used before.

When you check the platform.txt you will see that
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} {upload.verify} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"
is ok (without "-with_bootloader" (as in 2.0.2) but
tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" -v {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i"
is not ok. The "tools.avrdude.program.pattern" must be changed.

I use a STK500 programmer and definitely can reproduce the error.

slabbi, 2020-12-09:
Yes, I use the MegaCore:stk500.

slabbi, 2020-12-09:
btw:
recipe.hooks.savehex.presavehex.1.pattern.windows="{runtime.platform.path}/scripts/delete_merged_output.bat" {build.export_merged_output} "{build.path}\{build.project_name}.with_bootloader.hex"
fails, because I do not get this file. So this line fails and so the build process stops.

When I remove the ".with_bootloader" it removes the correct file (of course). So this line must be disabled (removed).
Note: It was not present in 2.0.2 and has been added later.

The disassembly part is ok when you like an additional file in the sketch home. Otherwise the "copy" line should also be removed, the disassembly can be picked up in the temp folder.

MCUdude, 2020-12-09:
What IDE version are you using?

I just tried to compile a sketch on both my Mac and my Windows computer using IDE 1.8.13. I'm not able to reproduce the issue. A *.with_bootloader.hex is _always_ generated (in the temp folder, the path can be found if you look at the compiler output). Can you try to explain what I should do in order to reproduce it, and not to fix it?

slabbi, 2020-12-10:
1.8.13
When you send me your email I can provide a Webex or Teams invitation if you want to check yourself ;)

slabbi, 2020-12-10:
This is from 2.0.5 (and 2.0.4), taken from the repo:

```
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} {upload.verify} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" -v {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i"

```
Do you see the difference between "program" (with ".with_bootloader" and "upload"? So "upload" should work, but "program" does not.

This is from 2.0.2 where it worked fine:

```
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} {upload.verify} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" -v {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"
```

Acc. the creation of both files. Yes, I remember that I got two files a long time ago when I used the bootloader upload, but didn't get this when I used the ISP upload for a long time now.

MCUdude, 2020-12-10:
> 1.8.13
When you send me your email I can provide a Webex or Teams invitation if you want to check yourself ;)

It's not that I don't believe you. It's just that I'd like to figure out what the root cause is; Why *.with_bootloader.hex files aren't created when you compile without bootloader. You see, *.with_bootloader.hex has a purpose. It lets you upload using your programmer without wiping the bootloader, as long as a bootloader option is selected in the Tools menu. All my cores use this file, and it has never been reported before.

Does the *.with_bootloader.hex file gets generated if you select Bootloader: Yes (UART0)?

slabbi, 2020-12-10:
No, when I check the logs I see that the compiler uses the option no_bootloader and uart0.
Nevertheless no ".with_bootloader file" is created. I used the patched files above.

```
board=2560
boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json,https://mcudude.github.io/MegaCore/package_MCUdude_MegaCore_index.json,https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json,https://mcudude.github.io/MicroCore/package_MCUdude_MicroCore_index.json,https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json,https://mcudude.github.io/MajorCore/package_MCUdude_MajorCore_index.json,https://mcudude.github.io/MegaCoreX/package_MCUdude_MegaCoreX_index.json
build.verbose=true
build.warn_data_percentage=75
builtin_platform_is_newer=10809
cache.enable=true
compiler.cache_core=true
compiler.warning_level=default
console=true
console.auto_clear=true
console.error.file=stderr.txt
console.length=500
console.lines=4
console.output.file=stdout.txt
custom_BOD=2560_2v7
custom_CrystalFreq=generic_26
custom_FlashFreq=generic_40
custom_FlashMode=generic_dout
custom_LTO=2560_Os_flto
custom_ResetMethod=generic_ck
custom_baud=generic_115200
custom_bootloader=2560_no_bootloader
custom_clock=2560_16MHz_external
custom_cpu=mega_atmega2560
custom_dbg=generic_Disabled
custom_eeprom=2560_keep
custom_eesz=generic_1M
custom_exception=generic_disabled
custom_ip=generic_lm2f
custom_led=generic_2
custom_lvl=generic_None____
custom_pinout=2560_avr_pinout
custom_sdk=generic_nonosdk221
custom_ssl=generic_all
custom_variant=328_modelP
custom_vt=generic_flash
custom_wipe=generic_none
custom_xtal=generic_80
editor.antialias=true
editor.auto_close_braces=true
editor.caret.blink=true
editor.code_folding=true
editor.divider.size=2
editor.external=false
editor.font=Monospaced,plain,13
editor.indent=true
editor.invalid=false
editor.keys.alternative_cut_copy_paste=true
editor.keys.home_and_end_beginning_end_of_doc=false
editor.keys.shift_backspace_is_delete=true
editor.languages.current=
editor.linenumbers=true
editor.save_on_verify=true
editor.tabs.expand=true
editor.tabs.size=2
editor.update_extension=true
editor.window.height.default=600
editor.window.height.min=290
editor.window.width.default=500
editor.window.width.min=400
export.applet.separate_jar_files=false
export.application.fullscreen=false
export.application.platform=true
export.application.stop=true
export.delete_target_folder=true
gui.scale=auto
ide.accessible=false
last.ide.1.8.10.daterun=1582303143
last.ide.1.8.10.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.ide.1.8.12.daterun=1592422518
last.ide.1.8.12.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.ide.1.8.13.daterun=1607599276
last.ide.1.8.13.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.ide.1.8.9.daterun=1570533569
last.ide.1.8.9.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.screen.height=1440
last.screen.width=2560
last.serial.location=-1108,124,1080,689
last.sketch.count=1
last.sketch.default.location=322,149,1520,1220,818,0
last.sketch.default.path=D:\Eigene Dateien\Arduino\xxx\xxx.ino
last.sketch0.location=322,149,1520,1220,818,0
last.sketch0.path=D:\Eigene Dateien\Arduino\xxx\xxx.ino
platform.auto_file_type_associations=true
preferences.readonly=false
preproc.color_datatype=true
preproc.enhanced_casting=true
preproc.imports.list=java.applet.*,java.awt.Dimension,java.awt.Frame,java.awt.event.MouseEvent,java.awt.event.KeyEvent,java.awt.event.FocusEvent,java.awt.Image,java.io.*,java.net.*,java.text.*,java.util.*,java.util.zip.*,java.util.regex.*
preproc.output_parse_tree=false
preproc.save_build_files=false
preproc.substitute_floats=true
preproc.substitute_unicode=true
preproc.web_colors=true
programmer=MegaCore:stk500
proxy.manual.hostname=
proxy.manual.password=
proxy.manual.port=
proxy.manual.type=HTTP
proxy.manual.username=
proxy.pac.url=
proxy.type=auto
recent.sketches=
run.display=1
run.options=
run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.present.bgcolor=#666666
run.present.exclusive=false
run.present.stop.color=#cccccc
serial.databits=8
serial.debug_rate=115200
serial.line_ending=2
serial.parity=N
serial.port=COM5
serial.port.file=COM5
serial.port.iserial=null
serial.show_timestamp=true
serial.stopbits=1
sketchbook.path=D:\Eigene Dateien\Arduino
software=ARDUINO
target_package=MegaCore
target_platform=avr
theme.file=
update.check=false
update.id=8319660653675244979
update.last=1565102282265
upload.using=bootloader
upload.verbose=true
upload.verify=true

```

slabbi, 2020-12-10:
Here is the output with the modified files (I shortened the output a little bit):

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xxx\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xxx\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Eigene Dateien\Arduino\libraries -fqbn=MegaCore:avr:2560:bootloader=uart0,eeprom=keep,pinout=avr_pinout,BOD=2v7,LTO=Os_flto,clock=16MHz_external -vid-pid=16C0_2AA9 -ide-version=10813 -build-path C:\Users\xxx\AppData\Local\Temp\arduino_build_479337 -warnings=default -build-cache C:\Users\xxx\AppData\Local\Temp\arduino_cache_563909 -prefs=build.warn_data_percentage=75 -verbose D:\Eigene Dateien\Arduino\xxx\xxx.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xxx\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xxx\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Eigene Dateien\Arduino\libraries -fqbn=MegaCore:avr:2560:bootloader=uart0,eeprom=keep,pinout=avr_pinout,BOD=2v7,LTO=Os_flto,clock=16MHz_external -vid-pid=16C0_2AA9 -ide-version=10813 -build-path C:\Users\xxx\AppData\Local\Temp\arduino_build_479337 -warnings=default -build-cache C:\Users\xxx\AppData\Local\Temp\arduino_cache_563909 -prefs=build.warn_data_percentage=75 -verbose D:\Eigene Dateien\Arduino\xxx\xxx.ino
Using board '2560' from platform in folder: C:\Users\xxx\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.5
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\xxx\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.5
Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\variants\\100-pin-avr" "-ID:\\Eigene Dateien\\Arduino\\libraries\\SSL-MegaGPIO-1.0" "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\libraries\\EEPROM\\src" "-ID:\\Eigene Dateien\\Arduino\\libraries\\SdFat\\src" "C:\\Users\\xxx\\AppData\\Local\\Temp\\arduino_build_479337\\sketch\\xxx.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\variants\\100-pin-avr" "-ID:\\Eigene Dateien\\Arduino\\libraries\\SSL-MegaGPIO-1.0" "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\libraries\\EEPROM\\src" "-ID:\\Eigene Dateien\\Arduino\\libraries\\SdFat\\src" "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\libraries\\SPI\\src" "C:\\Users\\xxx\\AppData\\Local\\Temp\\arduino_build_479337\\sketch\\xxx.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\variants\\100-pin-avr" "C:\\Users\\xxx\\AppData\\Local\\Temp\\arduino_build_479337\\sketch\\xxx.ino.cpp" -o "C:\\Users\\xxx\\AppData\\Local\\Temp\\arduino_build_479337\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\xxx\\AppData\\Local\\Temp\\arduino_build_479337\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Sketch wird kompiliert...
Compiling library "SPI"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\xxx\\AppData\\Local\\Arduino15 [... cut ...]
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\xxx\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.5\libraries\SPI  wird verwendet

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\xxx\\AppData\\Local\\Temp\\arduino_build_479337/xxx.ino.elf"
Der Sketch verwendet 126496 Bytes (48%) des Programmspeicherplatzes. Das Maximum sind 261120 Bytes.
Globale Variablen verwenden 5645 Bytes (68%) des dynamischen Speichers, 2547 Bytes für lokale Variablen verbleiben. Das Maximum sind 8192 Bytes.

```

slabbi, 2020-12-10:
I checked the MegaCore (with my patch) with a smaller code and I get two files now (with_bootloader and w/o).

Please note that my code is 255kb big, so there is no room left for a bootloader. I think the linker will not create the file because it is not possible.

MCUdude, 2020-12-10:
OK, I'm able to reproduce the issue. The large sketch size is the key here. It turns out that *.with_bootloader.hex isn't generated if the compiled binary is larger than 65536 bytes. This also affects MightyCore as well.

@per1234 is this a known issue with the IDE? That [sketchname.ino].with_bootloader.hex isn't generated for large sketches?

slabbi, 2020-12-10:
The ".with_bootloader" part is created by "\Arduino\arduino-builder.exe". So this is why I could not be able to find it somewhere else ;)

MCUdude, 2020-12-15:
The bug where *.with_bootloader.hex wasn't generated has been fixed in the latest hourly build. You can download this and give it a try. From my understanding, it won't be too long before IDE 1.8.14 is released. With this, I'm probably not going to do anything about this "bug", since it's really the IDE's fault, not MegaCore.

slabbi, 2020-12-15:
Agree, regarding the "Save HEX/Disassembly" part.

Nevertheless
tools.avrdude.upload.pattern=
AND
tools.avrdude.program.pattern=
should not have ".with_bootloader" there, as in the original Arduino platform.txt.

In yours you have added ".with_bootloader" in
tools.avrdude.program.pattern=

MCUdude, 2020-12-15:
> Nevertheless... should not have ".with_bootloader" there, as in the original Arduino platform.txt.

Yes, they should; I did explain this earlier. It's there because you can upload using a programmer and still not wipe the bootloader as you do on the official boards if you have selected a bootloader option in the Tools menu (UART0..UART3). 
If you select **No Bootloader** in the tools menu, it will bundle the [empty.hex](https://github.com/MCUdude/MegaCore/blob/master/avr/bootloaders/empty/empty.hex) file, which will result in a file identical to the one without the .with_bootloader suffix.

MCUdude, 2021-07-21:
This IDE bug has been fixed in the latest 1.8.15 IDE version.

closed 2021-07-21 by MCUdude