MCUdude/MegaCore #150: Uploading code through UART with Pololu USB AVR Programmer V2

sebashb, 2020-12-31:
I have a AT90CAN128 that is working fine, but I'm only able to upload code to it using the ISP port of the Pololu AVR Programmer. I have tried to use PlatformIO with the suggested platformio.ini file but I get the following error:
```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x31
```

This is my platformio.ini environment:
```
; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
;  Serial bootloader protocol
upload_protocol = arduino
; Serial upload port
upload_port = /dev/cu.usbmodem001703434
; Get upload baud rate defined in the fuses_bootloader environment
board_upload.speed = ${env:fuses_bootloader.board_bootloader.speed}

; Run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; Run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader
[env:fuses_bootloader]
board_hardware.oscillator = external ; Oscillator type
board_hardware.uart = uart0          ; Set UART to use for serial upload
board_bootloader.speed = 115200      ; Set bootloader baud rate
board_hardware.bod = 4.0v            ; Set brown-out detection
board_hardware.eesave = yes          ; Preserve EEPROM when uploading using programmer
board_hardware.jtagen = yes          ; Enable the JTAG port
board_hardware.ckout = yes           ; Enable clock output on PC7
upload_protocol = stk500             ; Use the stk500 as programmer
upload_flags =                       ; Select USB as upload port and divide the SPI clock by 8
  -P$UPLOAD_PORT
  -B8

[env:fuses_bootloader_MACOS]
extends = env:fuses_bootloader
upload_port = /dev/cu.usbmodem001703432

[env:fuses_bootloader_WIN]
extends = env:fuses_bootloader
upload_port = COM3
```

And this is my Pololu USB AVR programmer settings:
[image]

Am I missing something?

Thanks for the great work, it's awesome to be able to use a cheap AVR chip with CAN on PlatformIO!

MCUdude, 2020-12-31:
Hi!

Your hardware setup looks decent, so my guess would be that the Pololu programmer in "UART mode" doesn't toggle the DTR exactly like Avrdude/the bootloader does.

I do own an AT90CAN128 board, and I've never had any UART upload issues.

So the first question, have you loaded the bootloader by running `pio run -t bootloader -e fuses_bootloader`?
If yes, does pin PB5 LED toggle a few times if you press the reset button? This indicates that a bootloader is present.
Oh, are you able to print to the serial monitor in your program?

> Thanks for the great work, it's awesome to be able to use a cheap AVR chip with CAN on PlatformIO!

You're welcome! 🙂

sebashb, 2021-01-01:
Hi!

Thank you for your quick reply. Yes I have been able to upload the bootloader both with the command you referred and also with Arduino IDE, and yes after uploading the bootloader the LED starts blinking two times per second.

Serial monitor also works fine with the UART part of the Pololu programmer.

MCUdude, 2021-01-01:
Does the LED flash twice when you're trying to upload using UART? If so, the DTR toggles, which is supposed to do. If not, there's where the issue is.

sebashb, 2021-01-01:
I really don't know what I made different this time, but I have just managed to upload code through UART! I have reburned the bootloader and then I was able to upload code through UART. I have not changed any of the settings in platformio.ini since I have last tried to do it. I wish I could be more helpful and let you know what have changed but I really have no clue.

sebashb, 2021-01-01:
Just tried with a second board and it also worked, my guess is that I haven't burned the bootloader through Platformio and there was a different configuration when I uploaded it through Arduino IDE.

closed 2021-01-02 by sebashb