MCUdude/MegaCore #151: AT90CAN128 stucked in Wire.endTransmission()

sebashb, 2021-01-02:
Hi,

I'm trying to use the AT90CAN128 to read data from a INA226 with peterus/INA226Lib library but my code is getting blocked in the `ina.configure()` more precisely in the `writeRegister16(INA226_REG_CONFIG, config);` `Wire.endTransmission();` call. I have checked `twi.c` and chip is blocked in the while in line 277:

```
// wait for write operation to complete
  while(wait && (TWI_MTX == twi_state)){
    continue;
  }
```

The library should not be the problem since I have already used it before with a Teensy 3.2.

I'm using the Simple.ino example of the library, and I have already checked my connection between the INA and the AT90. Schematic is below:

[image]

[image]

sebashb, 2021-01-17:
@MCUdude  I have just tried the example `MegaCore/avr/libraries/Wire/examples/i2c_scanner/` and the same thing happened, the AT90CAN128 gets sucked on the `Wire.endTransmission();` have you tried I2C before on this chip?

MCUdude, 2021-01-26:
I just hooked up an i2c LCD 1602 driver, and I'm able to "ping" the i2c bus with an AT90CAN128. Note that `Wire` is blocking, so the program will hang if either the SDA or SCL line is held low.

Output:
```
I2C Scanner
Scanning...
I2C device found at address 0x20  !
done

Scanning...
I2C device found at address 0x20  !
done

Scanning...
I2C device found at address 0x20  !
done

...
```

sebashb, 2021-01-26:
Hi MCUdude, meanwhile I have desolder the INA226 and I was also able to ping the I2C bus but since I had no devices on the bus I have not closed this issue. I will test with a new INA during this week.

sebashb, 2021-01-31:
I have had the opportunity to solder a new INA226 to the PCB and now it's working the way it should. Sorry for the trouble.

closed 2021-01-31 by sebashb