MCUdude/MegaCore #153: UART (on Pins   PD2/PD3) on ATmega128

leonclear, 2021-01-14:
I have some code running on a Mega2560 pro board that I 

wanted to port to a Mega128 using Megacore to further 

integrate it in a HW design I have.

1. How to use the second (hardware-)UART (on Pins 

PD2/PD3) on ATmega128 ? show an example please.

how to access the port
can i use the code?:

`Serial2.begin(9600); 
int thisByte = 33; 
int secondByte = 37;
Serial2.print(", dec: "); 
Serial2.println(thisByte+secondByte);`

2. show an example for use attachInterrupt();
attachInterrupt(0). for  pin pe2

attachInterrupt(1). for  pin pe3

3. will the libraries work:

[https://www.arduinolibraries.info/libraries/one-wire](url)
[https://www.arduinolibraries.info/libraries/dallas-temperature](url)

to connect ds18b20. 

4. is it pwm pin : 12,13,14,15, 4, 5 ,53 ?

5. will the function analogWrite(12); work correctly? 

show an example for use  analogWrite(12);

MCUdude, 2021-01-14:
This sounds like a student assignment to me. Why don't you just try it and see if it works?

> How to use the second (hardware-)UART (on Pins
PD2/PD3) on ATmega128 ? show an example please.

You can use the hardware serial pins on pin PD2 and PD3. This is Hardware serial 1, and you use `Serial1.begin();`.

> show an example for use attachInterrupt(); attachInterrupt(0). for pin pe2 attachInterrupt(1). for pin pe3

Pin PE2 does not have interrupt capabilities. Neither does PE3. See pinout picture below. 

> will the libraries work...

Yes, but not on pin PF0 - PF7. See #102 for details

> is it pwm pin : 12,13,14,15, 4, 5 ,53 ?

Pin 53 is not available on ATmega128. There are PWN capabilities on pin 3, 4, 5, 12, 13, 14 and 15.

> will the function analogWrite(12); work correctly?

No, the ATmega128 only has 8 analog inputs.

[image]

closed 2021-01-14 by MCUdude

tomaskovacik, 2021-01-14:
> > will the function analogWrite(12); work correctly?
> 
> No, the ATmega128 only has 8 analog inputs.
analogWrite is output (PWM), no?

MCUdude, 2021-01-14:
My bad.

> will the function analogWrite(12); work correctly?

Yes, this will work since pin 12 has PWM capabilities

leonclear, 2021-01-15:
Please , show an example for use `attachInterrupt();`  for some pin.  Is function `attachInterrupt(4);`  for pin PE4  is correct?

tomaskovacik, 2021-01-15:
https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/

MCUdude, 2021-01-15:
> Please , show an example for use attachInterrupt();

It isn't my job to demonstrate standard Arduino functionality. You can read about interrupts here:
https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/

>  Is function attachInterrupt(4); for pin PE4 is correct?

Yes, that will work, but it is recommended to use `attachInterrupt(digitalPinToInterrupt(4), interruptFunction, mode)`