MCUdude/MegaCore #154: copy_flash_pages

admindotnu, 2021-01-16:
Quick question, and googling for some time now without finding a proper yes or no.

Is in the current megacore support for copy_flash_pages. And for current i mean installing the boards through boardmanager and without compiling myself.

Thank you !

MCUdude, 2021-01-16:
Hi! The pre-compiled Optiboot flash bootloaders do not support copy_flash_pages by default. for this, you'll have to compile your own bootloader.

MCUdude, 2021-01-16:
Ignore the message above. All targets with 64kB flash or more support copy_flash_pages by default. Sorry for not being clear, I had to look in the source code to be sure!

admindotnu, 2021-01-16:
Thank you for quick reply! 

I am going to dig deeper why me fetched bin is not applied.

closed 2021-02-07 by MCUdude

jandrassy, 2021-02-07:
@MCUdude Hans, makeall doesn't have COPY_FLASH_PAGES=1 for any MCU

MCUdude, 2021-02-07:
Sorry, I got it wrong, @jandrassy is right.

jandrassy, 2021-02-07:
@admindotnu, for which board you want Optiboot with my copy_flash_pages function?