MCUdude/MegaCore #156: MegaCore AVR Pinout Image has wrong labels

ilovetogetspamed, 2021-03-04:
Your MegaCore AVR Pinout image has the wrong port numbers for Arduino Pins 2-7.  There are other errors I'm sure but I didn't do a complete scan.

Arduino Pin 2 is Port E, pin 4. (PE4)
Arduino Pin 3 is Port E, pin 5. (PE5)
Arduino Pin 4 is Port G, pin 5. (PG5)
Arduino Pin 5 is Port E, pin 4. (PE3)
Arduino Pin 6 is Port E, pin 4. (PH3)
Arduino Pin 7 is Port E, pin 4. (PH4)

Thanks for wasting my time yesterday chasing this down.

MCUdude, 2021-03-04:
> Thanks for wasting my time yesterday chasing this down.

Really? You _blame_ me for this? This is an open-source project that I've created in my spare time, and I don't make any money out of it. Shame on you for using a 3rd party core that may have bugs.

With that being said, you've provided little information for me to go at, and what you have provided is a bit ambiguous. I assume you're referring to the **AVR 100-pin pinout** image? I can't find any issues with this or with the **Arduino MEGA** pinout. I know the MEGA pinout is a bit all over the place, but this is how the official pinout actually is.

Here's a small table that shows Arduino pin 2 to 7 for the two 100 pin pinouts.
Which of these are wrong?

Arduino pin no. | 100-pin AVR pinout | 100-pin Arduino MEGA pinout |
|-----------------|-----------------------|-----------------------------------|
| 2                      | PE2 (phy. pin 4)        |  PE4 (phy. pin 6)                        |
| 3                      | PE3 (phy. pin 5)        |  PE5 (phy. pin 7)                        |
| 4                      | PE4 (phy. pin 6)        |  PG5 (phy. pin 1)                       |
| 5                      | PE5 (phy. pin 7)        |  PE3 (phy. pin 5)                        |
| 6                      | PE6 (phy. pin 8)        |  PH3 (phy. pin 15)                     |
| 7                      | PE7 (phy. pin 9)        |  PH4 (phy. pin 16)                     |

Can you try to elaborate on what you think is wrong with the pinout you're referring to?

ilovetogetspamed, 2021-03-05:
Hans,

The physical pins don't match the Arduino pins as I outlined on my ticket.  I.e. on the 'MegaCore AVR pinout' image (which is the one that caused me all the wasted time), you list Arduino PIN 2 as bing on PE2 Port Pin when in fact it's on PE4 which is stated correctly on the "MegaCore Arduino pinout' image.

https://camo.githubusercontent.com/455bee6022ee397a2955a9d511bb8344f2bac04d1d437dd597b57a99affcf481/687474703a2f2f692e696d6775722e636f6d2f446652376172442e6a7067 <https://camo.githubusercontent.com/455bee6022ee397a2955a9d511bb8344f2bac04d1d437dd597b57a99affcf481/687474703a2f2f692e696d6775722e636f6d2f446652376172442e6a7067> has incorrect values, and does not match 

https://camo.githubusercontent.com/90c3301e9f9d2e07269f15f89ee823ea1ae3dca96e9b7d3e73a6a4d8d626c2b8/68747470733a2f2f692e696d6775722e636f6d2f4f37577457416a2e6a7067 <https://camo.githubusercontent.com/90c3301e9f9d2e07269f15f89ee823ea1ae3dca96e9b7d3e73a6a4d8d626c2b8/68747470733a2f2f692e696d6775722e636f6d2f4f37577457416a2e6a7067> which has correct values.

Unfortunately I was trying to work with the invalid one which is why I opened the ticket.

Anyhow, now you know, and can update your diagram if you care to.    I could have left you unaware and so some other unsuspecting soul could confused by it.

Sorry that I came across so harshly, but like I said, using the flawed github diagram wasted my time.  Time is one thing we can never get back.

Chris

MCUdude, 2021-03-05:
Even though the AVR pinout and the Arduino MEGA pinout are different, there is nothing wrong with any of them. Physical pins and Arduino pins isn't at all the same, and never has been. 

If you set Arduino pin 2 high by adding digitalWrite(2, HIGH); to your program, PE2 will go high if you're using the AVR pinout, and pin PE4 will go high if you're using the Arduino MEGA pinout.

tomaskovacik, 2021-03-05:
megacore support 3 pinouts, it is YOUR problem Chris that you did not educate your self, and do not select proper pinout. 

100pin avr pinout:
arduino 2 is Port PE:
https://github.com/MCUdude/MegaCore/blob/75cc3dbb1e0d9f1ce307037091d10a22204d087d/avr/variants/100-pin-avr/pins_arduino.h#L238
arduino 2 is bit 2 (so PE2):
https://github.com/MCUdude/MegaCore/blob/75cc3dbb1e0d9f1ce307037091d10a22204d087d/avr/variants/100-pin-avr/pins_arduino.h#L238

100in arduino pinout:
arduino 2 is Port PE:
https://github.com/MCUdude/MegaCore/blob/75cc3dbb1e0d9f1ce307037091d10a22204d087d/avr/variants/100-pin-arduino-mega/pins_arduino.h#L238
arduino 2 is bit 4:
https://github.com/MCUdude/MegaCore/blob/75cc3dbb1e0d9f1ce307037091d10a22204d087d/avr/variants/100-pin-arduino-mega/pins_arduino.h#L329

ilovetogetspamed, 2021-03-07:
From your MegaCore (incorrect Arduino pin labels):

From MegaCore Arduino (correct Arduino pin labels):

Mega Schematic from Arduino.cc <http://arduino.cc/> (https://www.arduino.cc/en/uploads/Main/arduino-mega2560-schematic.pdf <https://www.arduino.cc/en/uploads/Main/arduino-mega2560-schematic.pdf>):

PE4 is on PWML pin 2, PE5 is on PWML pin 3.  Your pinout image for MegaCore doesn't match the physical layout of the Arduino 2560 pin placement. i.e. THE BLUE ARDUINO PIN LABELS. If I'm wrong then please explain to me how  you can go from Arduino PWML pin 2 to PE2 pin 4 on on the MegaCore , and Arduino PWML pin 3 to PE3 pin 5 to the MegaCore Arduino?   Arduino's pin 2 will always connect to PE4 and pin 3 will always connect to PE5? The Arduino Mega2560 uses one chip, and one board layout.  If you look at the Microchip(Atmel) datasheet (http://ww1.microchip.com/downloads/en/DeviceDoc/ATmega640-1280-1281-2560-2561-Datasheet-DS40002211A.pdf <http://ww1.microchip.com/downloads/en/DeviceDoc/ATmega640-1280-1281-2560-2561-Datasheet-DS40002211A.pdf>) for the ATmega640-1281-2560-2561 the Physical Pin Numbers and Port Labels do not vary for the chips.
  
Not sure how I could have been clearer?  I should have consulted Arduino's schematic first.

You can twist perception, reality won't budge.

MCUdude, 2021-03-07:
> You can twist perception, reality won't budge.

I'm not trying to twist perception. I'm _really_ just trying to understand what you mean is wrong.

> PE4 is on PWML pin 2, PE5 is on PWML pin 3.  Your pinout image for MegaCore doesn't match the physical layout of the Arduino 2560 pin placement. i.e. THE BLUE ARDUINO PIN LABELS

First, let's just rule out ATmega2560 (100-pin) physical pins entirely. That doesn't matter _at all_. Port pins and "Arduino pins" don't line up with the physical pin number, so it doesn't make sense to talk about these anyways. The blue labels, "Arduino pin numbers" don't line up with the 100-pin layout no matter what pinout is in use.

>  If I'm wrong then please explain to me how you can go from Arduino PWML pin 2 to PE2 pin 4 on on the MegaCore , and Arduino PWML pin 3 to PE3 pin 5 to the MegaCore Arduino?

Because there are two pin mappings you can choose from, "MegaCore Arduino MEGA pinout" and "MegaCore AVR 100-pin" pinout. These represent two different pins_arduino files. It does not make sense to use the latter on an Arduino MEGA board, because the Arduino pin numbers won't match the board layout won't match the Arduino board layout. The "MegaCore AVR 100-pin" pinout is **intended for custom hardware where a more consistent pinout is preferred**. [This is the pins_arduino.h file for the 100-pin AVR pinout](https://github.com/MCUdude/MegaCore/blob/master/avr/variants/100-pin-avr/pins_arduino.h) and [this is the pins_arduino.h file for the Arduino MEGA pinout](https://github.com/MCUdude/MegaCore/blob/master/avr/variants/100-pin-arduino-mega/pins_arduino.h). It's nothing magical, it's just remapping of the pin numbers.

As for your example, PE3 is a pin that has PWM capabilities. This exact pin is mapped to Arduino in 5 in the "Arduino MEGA pinout", but is mapped to Arduino pin 3 in the "AVR 100-pin pinout". It's just a pure coincidence that digital pin 5 maps to physical pin 5 on the Arduino MEGA pinout. Pretty much no other pins do. The Arduino pin number is, just a number. I can make pin PE3 Arduino pin 99 if I want.

> Arduino's pin 2 will always connect to PE4 and pin 3 will always connect to PE5?

On the Arduino MEGA board, this is correct. But on a board designed around the "AVR 100-pin pinout", this is wrong.

ilovetogetspamed, 2021-03-07:
Ok, I understand that,  the physical pin numbers don't matter, it depends
on what header you include.   The easiest solution to avoid the confusion
would be to remove the arduino pin numbers from the MegaCore image since it
doesn't match up with the physical layout of the board.  That way no one
that sees that image would confuse it with the other arduino friendly one.

Anyhow, I came here for abuse, not an argument ;-) (from Monty Python's
Flying Circus,  incase you are unfamiliar)

Keep up the good work,  you've put a lot of effort into this project.   I
appreciate your taking the time to understand what I was having trouble
with,  hopefully we are clear now.

tomaskovacik, 2021-03-18:
> The easiest solution to avoid the confusion would be to remove the arduino pin numbers from the MegaCore image since it doesn't match up with the physical layout of the board.

are you saying this is not correct?
[image]

ilovetogetspamed, 2021-03-19:
Gotcha.  I see you are one of those guys that always needs to win.  So, you win.  It's your document, it's your github page.  Do whatever you want to.

MCUdude, 2021-03-19:
>  I see you are one of those guys that always needs to win.  So, you win.  It's your document, it's your github page.  Do whatever you want to.

It's not his Github repo, it's mine. Don't blame him. This isn't an argument, and no one is winning or losing. This issue didn't turn out to be a bug, it was a misunderstanding from your side, and there's nothing wrong with that.

closed 2021-03-19 by MCUdude