MCUdude/MegaCore #157: DIgital pin has some bug(?)

taoybb, 2021-03-30:
I tried to write a simple program and it seem to have a bug (?)

```
void setup() {
  pinMode(44, OUTPUT);

}

void loop() {
  digitalWrite(44, HIGH);
  delay(1000);
  digitalWrite(44, LOW);
}
```
Pin 44 will go HIGH with never going LOW as I coded, but if I add more code like below

```
void setup() {
  pinMode(44, OUTPUT);
  pinMode(43, OUTPUT);

}

void loop() {
  digitalWrite(44, HIGH);
  delay(1000);
  digitalWrite(44, LOW);
  digitalWrite(43, HIGH);
  delay(1000);
  digitalWrite(43, LOW);
}
```

Now it working as intended. Is it a bug or something? I'm using IDE version 1.8.9

tomaskovacik, 2021-03-30:
this will works for you;)

void loop() {
  digitalWrite(44, HIGH);
  delay(1000);
  digitalWrite(44, LOW);
  delay(1000);
}
;)

in your code   digitalWrite(44, HIGH);  is fired in  few microseconds after   digitalWrite(44, LOW); so you do not see it with naked eye

MCUdude, 2021-05-05:
@tomaskovacik is right!

closed 2021-05-05 by MCUdude