MCUdude/MegaCore #158: Failure to upload bootloader to ATmega2561

mrdesignr, 2021-04-08:
I'm using the arduino uno as the ISP connected according to your directions.  i get the messages, 
avrdude: reading input file "b11010110"
avrdude: invalid byte value (b11010110) specified for immediate mode
avrdude: read from file 'b11010110' failed

I can upload the bootloader successfully to the Atmega1281 and 328 successfully. I can upload further information about my procedures. But I just want to see if you know of this already from what i've said so far. Please advise. Thanks.

MCUdude, 2021-04-08:
Are you using the latest MegaCoreX version, v2.1.1? This bug was fixed in February, and v2.1.0 and v2.1.1 has this issue fixed

mrdesignr, 2021-04-08:
That was it! I realize now that arduino updates but does not necessarily install the updates. I went into the board manager and installed v2.1.1. My device, Atmega 2561, is not listed under MegaCoreX but under MegaCore. thx

MCUdude, 2021-04-08:
Great to hear you figured it out! MegaCoreX was just a typo, it's another Arduino core I'm currently doing some work on; MegaCore was what I meant to write.

closed 2021-04-08 by MCUdude