MCUdude/MegaCore #160: LowPower library

EarlVadim, 2021-04-15:
Hello.
I beg you to add a library for working with power consumption. 
I do not have enough knowledge to work with registers, so that AT90CAN sleeps 250-1000ms, wakes up, checks the CAN and, in the absence of a packet, falls asleep again.
For ATMega328 I was used LowPower library, but it not supported AT90CAN and not developed now.
Help me, please.

MCUdude, 2021-04-15:
This is something you'll have to contact the author of the library about.

However, it's not very difficult to use the AVR sleep library. Here's a guide for the ATmega328P. Shouldn't be very different.
https://www.gammon.com.au/forum/?id=11497

EarlVadim, 2021-04-15:
AT90 differs strongly... 
I don't have enough knowledge to change LowPower library self. And LP-library has not been developing for a long time. 
unlike yours MegaCore.

MCUdude, 2021-04-15:
Well, then you've painted yourself into a corner. You can always try the built-in Wiring functions. They provide some kind of sleep mode functionality. Haven't tried it with the CAN128 though.
https://github.com/MCUdude/MegaCore/blob/master/Wiring_reference.md#sleepmode

EarlVadim, 2021-04-15:
Thank you.. but..
The problem is not to put him to sleep (I can do this), but to make him wake up 1 time to 250-500ms, check the CAN and go back to sleep.

EarlVadim, 2021-04-15:
LowPower code for WDT
`
ISR (WDT_vect)
{
	// WDIE & WDIF is cleared in hardware upon entering this ISR
	wdt_disable();
}`

But AT90CAN havn't WDIE & WDIF ...  probably that's why I get a reset instead of waking up ... but how to wake up?

EarlVadim, 2021-04-15:
I make light PowerSave library  
https://github.com/EarlVadim/MegaCore/tree/master/avr/libraries/LowPower-AT90CAN

closed 2021-07-21 by MCUdude