MCUdude/MegaCore #162: Thoughts on why bootloader validation doesn't pass?

rkertesz, 2021-04-20:
Hello. Tried to burn the bootloader using avrdude and was getting through setting the fuses ok but when trying to copy and then **validate** the bootloader, I am getting an error. Having trouble understanding why it would happen. Using USBTinyISP, programming AVR128. Able to upload programs just fine but not sure why burn bootloader is an issue.

The specific error is
avrdude.exe: verification error, first mismatch at byte 0x1fc00
             0xff != 0x02

The terminal is shown below
```
PS C:\Users\x\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18\bin> .\avrdude.exe "-CC:\Users\x\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/avrdude.conf" -v -patmega128 -cusbtiny  "-Uflash:w:C:\Users\x\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/optiboot_flash/bootloaders/atmega128/7372800L/optiboot_flash_atmega128_UART0_115200_7372800L_B5_BIGBOOT.hex:i" -Ulock:w:0x0f:m

avrdude.exe: Version 6.3-20201216
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\x\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbtiny
avrdude.exe: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
             AVR Part                      : ATmega128
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
               flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : USBtiny
             Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude.exe: programmer operation not supported

avrdude.exe: Using SCK period of 10 usec
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.11s

avrdude.exe: Device signature = 0x1e9702 (probably m128)
avrdude.exe: safemode: lfuse reads as FF
avrdude.exe: safemode: hfuse reads as CE
avrdude.exe: safemode: efuse reads as FF
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: Using SCK period of 10 usec
avrdude.exe: reading input file "C:\Users\x\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/optiboot_flash/bootloaders/atmega128/7372800L/optiboot_flash_atmega128_UART0_115200_7372800L_B5_BIGBOOT.hex"
avrdude.exe: writing flash (131072 bytes):

Writing | ################################################## | 100% 0.03s

avrdude.exe: 131072 bytes of flash written
avrdude.exe: verifying flash memory against C:\Users\x\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/optiboot_flash/bootloaders/atmega128/7372800L/optiboot_flash_atmega128_UART0_115200_7372800L_B5_BIGBOOT.hex:
avrdude.exe: load data flash data from input file C:\Users\x\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/optiboot_flash/bootloaders/atmega128/7372800L/optiboot_flash_atmega128_UART0_115200_7372800L_B5_BIGBOOT.hex:
avrdude.exe: input file C:\Users\x\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/optiboot_flash/bootloaders/atmega128/7372800L/optiboot_flash_atmega128_UART0_115200_7372800L_B5_BIGBOOT.hex contains 131072 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.03s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x1fc00
             0xff != 0x02
avrdude.exe: verification error; content mismatch

avrdude.exe: safemode: lfuse reads as FF
avrdude.exe: safemode: hfuse reads as CE
avrdude.exe: safemode: efuse reads as FF
avrdude.exe: safemode: Fuses OK (E:FF, H:CE, L:FF)

avrdude.exe done.  Thank you.
```

tomaskovacik, 2021-04-20:
did you have anything on miso line other than the programmer? reading is ok so RST,CLK and MOSI lines are fine, writing is not => MISO is pulled high, or it is not pulled down by the programmer. also you stated that you were able to write fuses, did you try it immediately after this unsuccessful upload of bootloader?

rkertesz, 2021-04-20:
@tomaskovacik - There isn't anything strange connected to the MISO line as far as I can tell. I've programmed the device using a handheld ISP programmer with no problem so I'm a little stumped. 
Pardon my ignorance but how can you tell from the output that MISO is not low?
Regarding the fuse writing, it was immediately prior to the unsuccessful upload of the bootloader, as part of the operation that arduino "burn bootloader" performs. I have updated my initial post to include setting the fuses before sending the ihex.

MCUdude, 2021-04-20:
It is your programmer. The USBtinyISP can write past 64kiB, but can't read properly. Try with a different programmer and see for yourself 😊

tomaskovacik, 2021-04-20:
> @tomaskovacik - There isn't anything strange connected to the MISO line as far as I can tell. I've programmed the device using a handheld ISP with no problem before so I'm a little stumped.
> Pardon my ignorance but how can you tell from the output that MISO is not low?
> Regarding the fuse writing, it was immediately prior to the unsuccessful upload of the bootloader, as part of the operation that arduino "burn bootloader" performs. I have updated my initial post to include setting the fuses before sending the ihex.
> 
> I might try loading the ihex from the handheld ISP and see if it indicates any failures

with log from writing fuses, it is clear my theory was wrong, just for knowledge: 0xFF => MISO is high all time

rkertesz, 2021-04-20:
ah. I see. thank you both. 
I suppose that is what I get for using a cheapo ISP. 
Other than my handheld ISP and the USBtinyISP, all I have is an old USBASP which never seemed to work anyway.
I suppose there might be an argument I can use to skip verification but that seems a little risky.
My handheld programmed the chip without any error codes.
Thank you 👍

closed 2021-04-20 by rkertesz