MCUdude/MegaCore #164: It doesn't upload the bootloader during flash

D337z, 2021-05-07:
Hi!  So I'm trying to burn a bootloader to my ATMega2560 board and it's not actually copying anything to the bootloader area.  It claims to, but reading back the bootloader through serial shows nothing but 0xFF across the board and the board flashes fast twice every second.  The burn takes about 3 seconds, so I know it's not sending data.  I believe it's skipping straight to setting the fuses without actually sending the hex data.  I'm trying to use my Arduino Uno as a programmer, but it states that it's using the stk500v1.  There's definitely an issue there.
Here's the output of the command:

`avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\cryo_\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/avrdude.conf"

         Using Port                    : COM12
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xfd"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xfd:
avrdude: load data efuse data from input file 0xfd:
avrdude: input file 0xfd contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0b11010110"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0b11010110:
avrdude: load data hfuse data from input file 0b11010110:
avrdude: input file 0b11010110 contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xf7"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xf7:
avrdude: load data lfuse data from input file 0xf7:
avrdude: input file 0xf7 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified

avrdude done.  Thank you.

`

MCUdude, 2021-05-07:
So after you've burned the bootloader to the ATmega2560 you're not able to upload anything to it using a USB to serial adapter?

"Burn bootloader" consists of two Avrdude commands. The first Avrdude command sets the fuses, like in the Avrdude output you've posted, and the second Avrdude command writes the bootloader.

I'll recommend getting a dedicated programmer, such as the USBasp. It's always nice to have a programmer that _always_ works. There are usually lots of issues with Arduino as ISP. It doesn't mean it's something wrong with the code, but the whole wire-it-yourself thing is a significant source of error.

MCUdude, 2021-05-07:
Are you sure you've selected "Bootloader: Yes (UART0)"? and then ran "Burn Bootloader"?

Also, I highly recommend you upgrade to IDE 1.8.13. IIRC the 1.8.3 version has a lot of bugs that have later been fixed. I'm not 100% sure, but your issue _may_ be related to the IDE version.

D337z, 2021-05-07:
Yes. All settings are accounted for. And I actually am using the latest
version. I missed a 1.

MCUdude, 2021-05-07:
Can you turn on verbose compilation and upload in the IDE settings and post the entire output?

I just successfully burned the bootloader to an Arduino MEGA 2560 using a USBasp programmer, and I'm able to upload to it just fine

D337z, 2021-05-07:
Simple Blink example

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\cryo_\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\cryo_\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\cryo_\Documents\Arduino\libraries -fqbn=MegaCore:avr:2560:bootloader=uart0,eeprom=keep,pinout=mega_pinout,BOD=2v7,LTO=Os_flto,clock=16MHz_external -vid-pid=0403_6001 -ide-version=10813 -build-path C:\Users\cryo_\AppData\Local\Temp\arduino_build_456159 -warnings=all -build-cache C:\Users\cryo_\AppData\Local\Temp\arduino_cache_747444 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\cryo_\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\cryo_\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\cryo_\Documents\Arduino\libraries -fqbn=MegaCore:avr:2560:bootloader=uart0,eeprom=keep,pinout=mega_pinout,BOD=2v7,LTO=Os_flto,clock=16MHz_external -vid-pid=0403_6001 -ide-version=10813 -build-path C:\Users\cryo_\AppData\Local\Temp\arduino_build_456159 -warnings=all -build-cache C:\Users\cryo_\AppData\Local\Temp\arduino_cache_747444 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\cryo_\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
Using board '2560' from platform in folder: C:\Users\cryo_\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\cryo_\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2
Detecting libraries used...
"C:\\Users\\cryo_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\cryo_\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\cryo_\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.2\\variants\\100-pin-arduino-mega" "C:\\Users\\cryo_\\AppData\\Local\\Temp\\arduino_build_456159\\sketch\\Blink.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\Users\\cryo_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\cryo_\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\cryo_\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.2\\variants\\100-pin-arduino-mega" "C:\\Users\\cryo_\\AppData\\Local\\Temp\\arduino_build_456159\\sketch\\Blink.ino.cpp" -o "C:\\Users\\cryo_\\AppData\\Local\\Temp\\arduino_build_456159\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\cryo_\\AppData\\Local\\Temp\\arduino_build_456159\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
[... cut ...]
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xc9
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
`

D337z, 2021-05-07:
I tried it again.  It worked on the 5th try.

D337z, 2021-05-07:
`Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.3.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.4
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Checking size .pio\build\mega2560\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [========= ]  87.4% (used 7163 bytes from 8192 bytes)
Flash: [==========]  97.4% (used 247228 bytes from 253952 bytes)
Configuring upload protocol...
AVAILABLE: wiring
CURRENT: upload_protocol = wiring
Looking for upload port...
Auto-detected: COM11
Uploading .pio\build\mega2560\firmware.hex
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

*** [upload] Error 1
======================================================================================================================================== [FAILED] Took 70.64 seconds ========================================================================================================================================
Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:01:10.637
=================================================================================================================================== 1 failed, 0 succeeded in 00:01:10.637 ===================================================================================================================================The terminal process "C:\Users\cryo_\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

tomaskovacik, 2021-05-08:
hans probably mean that you post verbose output from uploading bootloader, no?

MCUdude, 2021-05-08:
@D337z Forget about PlatformIO for now. If you were to use it, you can't use `megaatmega2560` and expect it to work with MegaCore. Read the [MegaCore PlatformIO.md](https://github.com/MCUdude/MegaCore/blob/master/PlatformIO.md) file first.

As @tomaskovacik points out, I need you to post the verbose output when you burh the bootloader. It would be nice if you could post a screenshot of the Tools menu so I can verify that there isn't anything obviously wrong with your settings.

Yash333code, 2021-07-24:
what is this verbose output.

MCUdude, 2022-12-25:
No reply from OP for a very long time.

closed 2022-12-25 by MCUdude