MCUdude/MegaCore #166: MegaCore boot options

arhoufiry, 2021-06-09:
Can someone please explain what are all the different options in bootloaders?
I am using 16MHz crystal with the Atmega128A.  Right now, I can only upload sketches through Avrdude (hex file + bootloader).
Any help would be appreciated.

MCUdude, 2021-06-09:
> Can someone please explain what are all the different options in bootloaders?

I'm not sure what you mean by this. The bootloader is just a small program that lets you program the chip over UART instead of ISP. The bootloader is compiled to run on a chip with a specific clock frequency, e.g 16 MHz and a specific baud rate, 115200 baud.

If you can upload hex files using Avrdude, that's great! What is it that is unclear? What is your main issue?

arhoufiry, 2021-06-09:
The bootloaders are in folders labeled 16000000L......  I assume that it's the clock frequency.
The inside each folder, there are multiple bootloader files.  Since the documentation does not describe the files, I also assumed that each file is based on the settings (UART0 or UART1, baudrate, etc).
My hardware is as follow:  atmega128A, 5V supply, 16MHz external crystal.
bootloader file used:  optiboot_flash_atmega128_UART0_115200_16000000L_BIGBOOT.hex
Fuse bits:  
L:  0xEF
H: 0xCF
E: 0xFF
LB: 0x3F

I can successfully use the USBasp to write fuse bits and upload the bootloader or the a compiled sketch using a simple blink LED with Serial.print.
The LED blinks at the expected programmed rate 1/second, but I do not see anything on UART0.  I am also unable to upload sketches using an FTDI usb-uart bridge.
There are no connections issues as I checked and re-checked multiple times, and also used the FTDI bridge on other arduino boards.
I am not sure if my fuse settings are wrong or something in the bootloader file I selected.

MCUdude, 2021-06-09:
The reason for all the folders and long names is that the user options in Arduino IDE will pick the right bootloader for you. It will also pick the correct fuse settings. All you have to do is to select the settings you want in Arduino IDE and hit "Burn Bootloader". If you do this, the fuses are guaranteed to be correct.

The fuses you're using aren't correct. A 0xCF low fuse doesn't let you use the bootloader. 0xC6 will work. But, I recommend you use Arduino IDE to get familiar first.

arhoufiry, 2021-06-09:
I am not able to use the Arduino IDE because it asks for a serial port.  I currently can only use the USBasp dongle to program through ISP.  I am using Averdudess gui, this is the only successful option for me at this time (except for the UART not working).

[image]

These are the messages I get when I try to upload the sketch using the USBasp (MegaCore also used):
Arduino: 1.8.13 (Windows 10), Board: "ATmega128, Yes (UART1), EEPROM retained, BOD disabled, LTO disabled, External 16 MHz"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Abdel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Abdel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Abdel\Documents\Arduino\libraries -fqbn=MegaCore:avr:128:bootloader=uart1,eeprom=keep,BOD=disabled,LTO=Os,clock=16MHz_external -ide-version=10813 -build-path C:\Users\Abdel\Documents\Arduino\arduino_builds -warnings=none -build-cache C:\Users\Abdel\AppData\Local\Temp\arduino_cache_101090 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\Abdel\Documents\Arduino\Blink_modified\Blink_modified.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Abdel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Abdel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Abdel\Documents\Arduino\libraries -fqbn=MegaCore:avr:128:bootloader=uart1,eeprom=keep,BOD=disabled,LTO=Os,clock=16MHz_external -ide-version=10813 -build-path C:\Users\Abdel\Documents\Arduino\arduino_builds -warnings=none -build-cache C:\Users\Abdel\AppData\Local\Temp\arduino_cache_101090 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Abdel\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\Abdel\Documents\Arduino\Blink_modified\Blink_modified.ino

Using board '128' from platform in folder: C:\Users\Abdel\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.1

Using core 'MCUdude_corefiles' from platform in folder: C:\Users\Abdel\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.1

Detecting libraries used...

"C:\\Users\\Abdel\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -fpermissive -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR "-IC:\\Users\\Abdel\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.1\\cores\\MCUdude_corefiles" "-IC:\\Users\\Abdel\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.1\\variants\\64-pin-avr" "C:\\Users\\Abdel\\Documents\\Arduino\\arduino_builds\\sketch\\Blink_modified.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"C:\\Users\\Abdel\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -fpermissive -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR "-IC:\\Users\\Abdel\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.1\\cores\\MCUdude_corefiles" "-IC:\\Users\\Abdel\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.1\\variants\\64-pin-avr" "C:\\Users\\Abdel\\Documents\\Arduino\\arduino_builds\\sketch\\Blink_modified.ino.cpp" -o "C:\\Users\\Abdel\\Documents\\Arduino\\arduino_builds\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LI [... cut ...]
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Abdel\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.1/avrdude.conf"

         Using Port                    : COM30

         Using Programmer              : arduino

         Overriding Baud Rate          : 115200

avrdude: ser_open(): can't open device "\\.\COM30": The system cannot find the file specified.

avrdude done.  Thank you.

MCUdude, 2021-06-09:
But in order to upload using the bootloader, you need a serial port. That's the thing with the bootloader! You can also use your USBasp programmer for uploading using Arduino IDE, but then you'll have to select "No bootloader" in the tools menu.

arhoufiry, 2021-06-09:
The atmega128 uses PDO/PDI that are multiplexed with TX0/RX0.  The thing that baffles me is that those lines are used to upload hex files, but do not work as a uart. I also tried a bootloader that enables UART1 with no success.
I revisited the fuse bits but with no luck:
Low Fuse:  0xEF
High Fuse: 0xD9
Extended: 0xFF
Lockbit: 0xFF
[image]

MCUdude, 2021-06-09:
> The atmega128 uses PDO/PDI that are multiplexed with TX0/RX0. The thing that baffles me is that those lines are used to upload hex files, but do not work as a uart. 

This is one of the quirky things with the ATmega64/128/1281/2561 series. However, I've never had any issues using the RX0 and TX0 pins. You need to disconnect the programmer before you use the UART interface though.

I'm not sure why you care so much about finding the right fuse values manually. I've designed MegaCore to do this for you. All you have to do is to select the options in the tools menu and hit "Burn bootloader". Even though the button says "Burn Bootloader" it can also be used to _only_ set the fuses if you select the no bootloader option. One issue with your fuse settings is that you haven't turned on the BOOTRST fuse. The boot flash section size should also be set to 512 words.

But no matter what you try, you won't be able to upload using the bootloader if you don't have a serial port available, usually a USB to serial adapter.

arhoufiry, 2021-06-09:
Right now, the arduino ide doesn't let me program anything.  I can only use Averdudess.
I tested with an arduino mega2560, loaded the blink sketch hex file with (optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex), and the uart0 does not work.  I did not touch the fuses on this.

MCUdude, 2021-06-09:
> Right now, the arduino ide doesn't let me program anything. I can only use Averdudess.

So let's focus on ATmega128 with Arduino IDE.

* Are you able to set the settings in the tools menu, select your USBasp programmer and hit "Burn bootloader"? Please turn on verbose output in the IDE settings and post the output you get?
* Now the bootloader should be loaded. Select the serial port you've going to use for uploading, and click the upload button. Please paste the output of this as well.

arhoufiry, 2021-06-09:
OK, I've made a little progress regarding the USBasp.
I used Zadiq.exe and changed the driver to "libusbK".  After that I am able to load the bootloader from the Arduino IDE, but I can only upload the sketch if the bootloader is not selected under "MegaCore" settings.  I can compile the blink sketch and generate a combined hex file, that I am able to upload using Averdudess.  Again the LED blinks at the correct rate but still no serial output.
I also tried what you suggested above and here are the outputs:
[bootloader_output.txt](https://github.com/MCUdude/MegaCore/files/6626970/bootloader_output.txt)
[serial_upload.txt](https://github.com/MCUdude/MegaCore/files/6626971/serial_upload.txt)

tomaskovacik, 2021-06-10:
do you have auto-reset capacitor installer? (100nF between DTR of serial adapter and reset pin of atmega)

MCUdude, 2021-06-10:
@arhoufiry There is probably something wrong with your auto reset circuity.

If you connect an LED to pin PB5, does it flash twice if you reset the chip within a second, either by pressing a button or shorting the pin to ground and releasing it? This means that a bootloader is present and working.

arhoufiry, 2021-06-10:
I get 2 blinks repeating every second with just the bootloader uploaded.  Here is my schematic:
[image]

arhoufiry, 2021-06-10:
There must be an issue with the MegaCore.  I tried the same process on an Arduino Mega2560.  The bootloader loads fine using USBasp, B5 LED blinks @ 2 /sec.  Tried setting both uart0 and uart1, and in both cases the FTDI serial programming fails.

arhoufiry, 2021-06-11:
I ended up populating a second board, and I can upload the sketches fine.  I have some work to do to figure out what's wrong with the first board.
I can post what I find out before closing this issue, if anyone is interested.

MCUdude, 2021-06-11:
> I can post what I find out before closing this issue, if anyone is interested.

Please do!

arhoufiry, 2021-06-21:
I found the issue with the FTDI usb-uart bridge.  The FTDI ic was a knockoff.
I am still having the issue with UART1 not working as a serial port, but I can use it to upload sketches if the uart1 bootloader is programmed.

MCUdude, 2021-07-21:
Since the issue seems to be resolved, I'm closing this issue.

closed 2021-07-21 by MCUdude