MCUdude/MegaCore #167: i have this error

fadi0albaghdadi, 2021-07-01:
Low memory available, stability problems may occur.
Does it affect the processor? atmega128

lshw, 2021-07-06:
The stack length may be insufficient.

tomaskovacik, 2021-07-06:
This is problem with your code not core. You must use micro with more ram or optimise yout code. Anyway we need more informations about your hw setup and your code

fadi0albaghdadi, 2021-07-07:
> The stack length may be insufficient.

Does this affect the operation of the processor?

fadi0albaghdadi, 2021-07-07:
> This is problem with your code not core. You must use micro with more ram or optimise yout code. Anyway we need more informations about your hw setup and your code

Does this affect the operation of the processor?

tomaskovacik, 2021-07-07:
50:50 chance

- try it
- google warning message and read what was already witen elsewhere
- we can help you only if you post your code here

fadi0albaghdadi, 2021-07-07:
> 50:50 chance
> 
> * try it
> * google warning message and read what was already witen elsewhere
> * we can help you only if you post your code here

The code is very private, it cannot be published and is too large

tomaskovacik, 2021-07-07:
Then you must simply test it

MCUdude, 2021-07-07:
Optimize your code to use less RAM, or choose a larger, pin compatible microcontroller with more RAM such as the Atmega1281 or ATmega2561.

closed 2021-07-07 by MCUdude

fadi0albaghdadi, 2021-07-07:
Does that mean I can replace the atmega128 processor with atmega1281

MCUdude, 2021-07-07:
Yes, as long as your hardware setup doesn't depend on anything unique to the ATmega128.

lshw, 2021-07-10:
use  F(string) to put string into ROM.

Serial.println("hello"); // use 5 byte ram,
Serial.println(F("hello")); // use 0 byte ram

F(string)   can save a lot of ram.