MCUdude/MegaCore #170: ATMEGA128A MCU programming

Yash333code, 2021-07-24:
I want to program my atmega128a microcontroller , the code is written on arduino IDE ,and i m making arduino uno or mega as a programmer (*1) but when i burn bootloader in it- it shows "signature not match" *2.
*1 - the microcontroller is mounted on the PCB. arduino uno or mega as a programmer and connecting (MISO-MOSI connection from arduino to MCU. 
*2 - I have install megacore library but it does not show "128a" but it do show "128" curently selecting 128. 

Issue - Iam unable to program atmega128a MCU.
Current Task - want to upload program with current setup. ie- arduino IDE, arduino uno or mega , on board programming .
but not limited to current setup.
thanks any help will be appreciatable

MCUdude, 2021-07-24:
Open IDE settings, turn on verbose upload and post the error message. Also note that the ATmega128 uses PDI and PDO for ISP upload, not MOSI and MISO.

[image]

Yash333code, 2021-07-24:
[image]

I have implemented PDI PDO  for ISP still showing this when burn bootloader (Note:- I am using atmega128a not atmega128).
*also "device signature changes every time i burn bootloader

error:-

C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\pp\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf -v -patmega128 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0b11000110:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\pp\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Error while burning bootloader.
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x80ff00
avrdude: Expected signature for ATmega128 is 1E 97 02
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

MCUdude, 2021-07-24:
The programmer cannot communicate with the ATmega128. It may be the crystal or the wiring.

Yash333code, 2021-07-24:
yes , for crystal i have checked with internal oscilator. result :- no improvement.
wiring :- i have done continuity test through out the board no issue found.
still a question in my mind - can we program atmega128a through this setup?

Yash333code, 2021-07-24:
verbose output(i hope so)
[image]
[image]
 :- 

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pp\AppData\Local\Arduino15\packages -hardware C:\Users\pp\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pp\Documents\Arduino\libraries -fqbn=MegaCore:avr:128:bootloader=uart0,eeprom=keep,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=2341_0043 -ide-version=10813 -build-path C:\Users\pp\AppData\Local\Temp\arduino_build_435476 -warnings=none -build-cache C:\Users\pp\AppData\Local\Temp\arduino_cache_592320 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\pp\Desktop\UV_Blink\UV_Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pp\AppData\Local\Arduino15\packages -hardware C:\Users\pp\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pp\Documents\Arduino\libraries -fqbn=MegaCore:avr:128:bootloader=uart0,eeprom=keep,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=2341_0043 -ide-version=10813 -build-path C:\Users\pp\AppData\Local\Temp\arduino_build_435476 -warnings=none -build-cache C:\Users\pp\AppData\Local\Temp\arduino_cache_592320 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\pp\Desktop\UV_Blink\UV_Blink.ino
Using board '128' from platform in folder: C:\Users\pp\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\pp\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3
Detecting libraries used...
"C:\\Users\\pp\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR "-IC:\\Users\\pp\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\pp\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.3\\variants\\64-pin-avr" "C:\\Users\\pp\\AppData\\Local\\Temp\\arduino_build_435476\\sketch\\UV_Blink.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\Users\\pp\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR "-IC:\\Users\\pp\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\pp\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.1.3\\variants\\64-pin-avr" "C:\\Users\\pp\\AppData\\Local\\Temp\\arduino_build_435476\\sketch\\UV_Blink.ino.cpp" -o "C:\\Users\\pp\\AppData\\Local\\Temp\\arduino_build_435476\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\pp\\AppData\\Local\\Temp\\arduino_build_435476\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
[... cut ...]
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

An error occurred while uploading the sketch
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0xff00ff
avrdude: Expected signature for ATmega128 is 1E 97 02
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

tomaskovacik, 2021-07-24:
You get different signature each time. I would remove crystal, change fuses to use internal oscilator if the chip has one and try again, if it help, crystal is problem. Basicaly try to use as minimal setup as posible. If it does not work with minimal setup: I would try another chip to confirm my programer works, if it works the chip is bad.

Yash333code, 2021-08-06:
OK, For now i have ordered the avr board(NEX Robotics)
[image]
[image]
 from the market with the same controller, programmed it , then desoldered it from the avr board and mounted on my hardware. Result :- working fine.
espected issue:- may be the controllers are faulty , still communicating with microchip for fresh controllers will update about the issue.

MCUdude, 2022-04-20:
I'm closing this issue because it's highly unlikely that this is an issue caused by MegaCore, but rather is a hardware problem

closed 2022-04-20 by MCUdude