MCUdude/MegaCore #172: Unable to Burn Bootloader to AT90CAN128 with ATMEL-ICE

syph3rd, 2021-07-30:
Trying to burn bootloader to the AT90 but it keeps failing when trying to process a lockfile.
Can Read and Write to the Chip with Atmel Studio 7.0 and erase etc fine.

vrdude: usbdev_open(): Found Atmel-ICE CMSIS-DAP, serno: J42700021700
avrdude: max packet size expected 912, but found 512 due to EP 0x01's wMaxPacketSize
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : AT90CAN128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     8    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65     6   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : Atmel-ICE (ARM/AVR) in ISP mode
         Vtarget         : 4.8 V
         SCK period      : 125.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9781 (probably c128)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xff instead of 0x3f (double check with your datasheet first).
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xff"
avrdude: writing efuse (1 bytes):

Error while burning bootloader.
Writing | ################################################## | 100% 0.09s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xff:
avrdude: load data efuse data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xf5 != 0xff
avrdude: verification error; content mismatch

avrdude done.  Thank you.

MCUdude, 2021-07-30:
What happens if you set the BOD level to disabled instead of 2.7V?

syph3rd, 2021-08-03:
C:\Users\syph3rd\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\syph3rd\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf -v -pat90can128 -catmelice_isp -Pusb -e -Ulock:w:0xff:m -Uefuse:w:0xff:m -Uhfuse:w:0b11011111:m -Ulfuse:w:0xff:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\syph3rd\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : atmelice_isp
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : AT90CAN128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     8    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65     6   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : Atmel-ICE (ARM/AVR) in ISP mode
         Vtarget         : 4.9 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9781 (probably c128)
avrdude: erasing chip
avrdude: reading input file "0xff"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xff:
avrdude: load data lock data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xff"
avrdude: writing efuse (1 bytes):

Error while burning bootloader.
Writing | ################################################## | 100% 0.09s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xff:
avrdude: load data efuse data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xf5 != 0xff
avrdude: verification error; content mismatch

avrdude done.  Thank you.

mcuee, 2022-07-23:
Just want to report that I can burn the bootloader for the AT90CANUSB (on an old DVK90CAN1-1.1 version). It is using 8MHz oscillator and UART0. 

1) burning the bootloader
```
C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/avrdude.conf -v -pat90can128 -cusbasp -Pusb -e -Ulock:w:0xff:m -Uefuse:w:0xf5:m -Uhfuse:w:0b11010110:m -Ulfuse:w:0xff:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : AT90CAN128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     8    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65     6   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9781 (probably c128)
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "0xff"
avrdude: writing lock (1 bytes):

C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/avrdude.conf -v -pat90can128 -cusbasp -Pusb -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/bootloaders/optiboot_flash/bootloaders/at90can128/8000000L/optiboot_flash_at90can128_UART0_57600_8000000L_B5_BIGBOOT.hex:i -Ulock:w:0xcf:m 
Writing |  ***failed;  
################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xff:
avrdude: load data lock data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
[... 191 lines cut ...]

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9781 (probably c128)
avrdude: reading input file "C:\Users\xiaof\AppData\Local\Temp\arduino_build_515350/Blink_Arduino.ino.hex"
avrdude: writing flash (1246 bytes):

Writing | ################################################## | 100% 0.40s

avrdude: 1246 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Temp\arduino_build_515350/Blink_Arduino.ino.hex:
avrdude: load data flash data from input file C:\Users\xiaof\AppData\Local\Temp\arduino_build_515350/Blink_Arduino.ino.hex:
avrdude: input file C:\Users\xiaof\AppData\Local\Temp\arduino_build_515350/Blink_Arduino.ino.hex contains 1246 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.38s

avrdude: verifying ...
avrdude: 1246 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-23:
@syph3rd 
Maybe you can also try to burn the bootloader using new version of avrdude 7.0 to see if that helps. It may be related to Atmel-ICE support.

Ref: latest beta release of Arduino avrdude 7.0 version
https://github.com/arduino/avrdude-packing/releases

syph3rd, 2022-08-31:
Hello, 

Sorry for not replying, Using avrdude 7.0 fixed this issue

mcuee, 2022-12-21:
In this case, this issue can be closed.