MCUdude/MegaCore #173: Need a Bootloader for ATmega2561 working on 12.500 MHz

miksumin, 2021-07-30:
I am using MegaCore with ATMega2561 working on External 12.500 MHz frequency but there is no Optiboot bootloader for it. Can you please build it for me?

MCUdude, 2021-07-30:
It's all there. Build it yourself, it's a good learning experience

miksumin, 2021-07-30:
I don't have 'make' package

miksumin, 2021-07-30:
There is no ..\..\..\tools\avr\utils\bin\make

miksumin, 2021-07-30:
Please help

miksumin, 2021-07-30:
make atmega2561 AVR_FREQ=12500000L BAUD_RATE=115200 LED=G0 LED_START_FLASHES=2 UART=0/1

MCUdude, 2021-07-30:
A baud rate of 115200 at 12.5 MHz wouldn't work at all. Too much error. Look at the U2X=1 field, that's what the bootloader uses. Choose a green baud rate instead.

[image]

miksumin, 2021-07-30:
Ok, I hope 57600 would be ok. So let's use the following:
make atmega2561 AVR_FREQ=12500000L BAUD_RATE=57600 LED=G0 LED_START_FLASHES=2 UART=0/1

MCUdude, 2021-07-30:
OK, here you go.

[12500000L.zip](https://github.com/MCUdude/MegaCore/files/6909342/12500000L.zip)

miksumin, 2021-07-30:
Many thanks

closed 2021-07-30 by MCUdude