MCUdude/MegaCore #175: Using copy_flash_pages

rkertesz, 2021-09-15:
Are there any basic examples or instructions on how to use copy_flash_pages to receive a hex over uart from another device ( like another Arduino) or from an ftp server and then write the new firmware to the boot section?

Is this already available in bigboot?

jandrassy, 2021-09-15:
https://github.com/jandrassy/ArduinoOTA/blob/master/examples/Advanced/OTASketchDownload/OTASketchDownload.ino
only you will read the binary from Serial, not from WiFiClient

rkertesz, 2021-09-15:
I see this statement in the comments at the top of the example but didn't find any instructions in the readme. is there something I need to do to the hex file?

>  To create a bin file for AVR boards see the instructions in README.MD.

jandrassy, 2021-09-15:
>To use remote upload from IDE with SDStorage or InternalStorage, copy platform.local.txt from extras/avr folder, next to platform.txt in the boards package used (Arduino-avr or MCUdude packages). 

the configuration files setup creation of a bin file

closed 2022-04-20 by MCUdude