MCUdude/MegaCore #176: ATmega128 Pin Configuration

csmith608, 2021-09-16:
I am using this bootloader for optiboot_flash_atmega128_UART0_38400_8000000L_BIGBOOT.hex.  

I was looking at the pin configuration for RXD0 pin and saw that PORTE0 is set to 0, meaning that the internal pull-up resistor is not activated. As I understand it, if the RX pin is left floating (as it is in my application) it could read garbage data and reset the controller.  The fix is to activate the internal pull-up resistor.  I was wondering why it wasn't activated in the bootloader and maybe how I could do it for my application.  

Referencing page 66 and 87 https://ww1.microchip.com/downloads/en/DeviceDoc/doc2467.pdf

MCUdude, 2021-09-16:
The bootloader is only actively running after getting a reset pulse. When the microcontroller is powered up, the bootloader starts, but immediately jumps to the user program without touching any registers.

When you're uploading a new program and bootloader is finished writing new content to the flash memory, it gets a new reset pulse from Avrdude. The USB to serial adapter is actively driving the RXD0 pin, and since it will start executing the user application within 1 second when no valid data is received, I see this as a non-issue.

csmith608, 2021-09-16:
I see, thank you for the explanation!

closed 2021-09-16 by csmith608