MCUdude/MegaCore #177: Burn Atmega128A with USBasp

muhit313, 2021-09-17:
Some days ago I bought one **Atmega128A** and one **USBasp** for new my project. Because I need more then UNO memory. But I am little bit confused about how to burn the bootloader of Atmega128A with USBasp. Because there is no schematic for USBasp. 
So I try it myself as:
**USBasp...............Atmega128A**
MISO>>>>>>>PE0(PDI)
MOSI>>>>>>>PE1(PDO)
SCK>>>>>>>>PB1(SCK)
RST>>>>>>>>RESET
VCC>>>>>>>>VCC
GND>>>>>>>GND

And then try to burn the bootloader using your MegaCore. But I am unsuccessful. It gives me error. So can anyone help me about the **Schematic for USBasp** and **how can I burn and program the Atmega128A chip with USBasp.**

**Note:** I also read other issue and most of the solution is using **USBtinyISP**. But i don't have USBtinyISP. So Please suggest me **Schematic** for USBasp.
I can manage **FT232RL** or **CP2102 USB  to TTL UART** Module. If there is any chance to work with them.

**Errror:**
```Arduino: 1.8.13 (Windows 10), Board: "ATmega128, Yes (UART0), EEPROM retained, BOD 2.7V, LTO disabled, Internal 8 MHz"

avrdude: Version 6.3-20201216

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Name\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : usb

         Using Programmer              : usbasp

         AVR Part                      : ATmega128

         Chip Erase delay              : 9000 us

         PAGEL                         : PD7

         BS2                           : PA0

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff

           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff

           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp

         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)

avrdude: error: program enable: target doesn't answer. 1 

avrdude: initialization failed, rc=-1

         Double check connections and try again, or use -F to override

         this check.

avrdude done.  Thank you.

Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

MCUdude, 2021-09-17:
The MOSI pin on the USBasp connects to the PDI pin, and MISO on the USBasp connects to the PDO pin. See the minimal setup schematic for 64-pin AVRs. If you're still not getting it to work, it may be because the USBasp programmer is too fast for the microcontroller. You can then use the "USBasp (slow)" option when you burn the bootloader.

What do you use as the clock source? A 16 MHz crystal or the internal oscillator? Is the chip brand new/factory reset, or has it been used for something else before?

[image]

muhit313, 2021-09-18:
**The minimal setup schematic for 64-pin AVRs is little bit confusing** for me because there is no **UART header in USBasp** . 
So how can I connect DTR,R,Tx etc?

I also use the "USBasp (slow)" option when you burn the bootloader. But no luck.

I use clock source as internal oscillator.
Yes the chip is brand new and not used before.

tomaskovacik, 2021-09-18:
obviously, if you do not have a serial interface, then do not connect it?

anyway, this works for me most of the time:
try set internal oscillator to 1MHz plus "USBasp (slow)" => burn bootloader  then change it to your desired speed with "USBasp (slow)"  and then is should work also with standard usbasp

muhit313, 2021-09-19:
Thanks a lot to all of you. After 2 days hard and soul try I can see the success. May be I try all of the possible way that I can. 
I think the chip is only used with 16Mz crystal. Because I am unable to use the 1/8KHz Internal crystal. If I want to use them then I will faces this kind of error. 
**So finally I put 16KHz crystal and select  USBasp slow(MegaCore) and its working fine.** 
Thanks again to all especially  @MCUdude for his support and this nice board.

MCUdude, 2021-09-19:
Great to hear you figured it out! What probably happened was that the chip was already set up to use an external crystal, and it then won't work without it, and you can't communicate with it using a programmer.

You can now select the internal oscillator in Arduino IDE and burn the bootloader again, but you can also continue to use it with a 16 MHz crystal.

closed 2021-09-19 by MCUdude

muhit313, 2021-09-19:
Yes you are right @MCUdude . Sorry for another comment.
Now, Can I use **usb to ttl** or **USB to Serial adapter** or **FTDI** to program and open serial communication?
Because my USBasp may be not support serial communication.