MCUdude/MegaCore #182: "Burn Bootloader" that can only be used once

13507086298, 2022-04-24:
I write the Atmega128 via "Burn Bootloader" and it works fine, but after I upload the sketch once via UART1, the Bootloader doesn't work anymore, which means "Burn Bootloader" is a one-off in my actual use. I would like to know how to keep the "Burn Bootloader". Allows me to upload sketches via UART1 each time
[image]
[image]
[image]
[image]
But I will go up the grass map again, there will be an error.
[image]
This indicates that Bootloader has failed. I can make UART1 get the sketch through UART1 by using Tools-> Burn Bootloader, and it will fail again after success.

tomaskovacik, 2022-04-24:
did you implemented auto-reset circuit?

13507086298, 2022-04-24:
@tomaskovacik  Thank you very much for your reminder, I have an Auto-reset circuit, but it doesn't seem to work, I can upload normally after I reset it manually, I will check my hardware circuit. thank you very much

closed 2022-04-24 by 13507086298