MCUdude/MegaCore #189: Same code and pcb, Atmega128 can use 'String' but Atmega169 empty

happysoul, 2022-06-14:
I use same code and pcb like this

`String hexChip(char* a){
    String s = "";
    for(int i=0;i<6;i++){
      if(a[i]!=0x20){ // 0x20 is ascII space
        s += String(a[i]);
      }
    }
    return s;
}`

char a[20] = "abc de fghi...";
String re = hexChip(a+4);

128 return "defgh"
169 return empty

169 not support "String" ?

MCUdude, 2022-06-14:
Hi!
Grat to see you're using the ATmega169! I _just_ added support for it 🙂 Are you using it because of the chip shortage? I'm asking because it tends no not be very popular.

It _should_ support `String` out of the box, like any Arduino. The following sketch outputs `defgh` on my setup. In other words, I can't reproduce the issue.

```cpp
String hexChip(char* a){
    String s = "";
    for(int i = 0; i < 6; i++)
    {
      if(a[i] != 0x20){ // 0x20 is ascII space
        s += String(a[i]);
      }
    }
    return s;
}

void setup() {
  Serial.begin(9600);
  char a[20] = "abc de fghi...";
  String re = hexChip(a+4);
  Serial.println(re);
}

void loop() {
}
```

I'm using an ATmega169PA. What chip (including the suffix) are you using?

happysoul, 2022-06-14:
ATmega169PA-AU
about 1 week，when i see the 169 support，i try to use 169 from board of 128
I got ATmega169PA-AU and ATmega128 from Alibaba(TaoBao) . 169-￥6.8, 128￥6 （about 1 doller, ）
But i think 169 is easy to get from internet shopping 
I try to make a eeprom-programmer to read-write 32pin chip, like mx29f040 windond29040
[image]
Connect : PC - usb - ch340 - atmega
PC send commend to board  : x 29010 00000 7fffff
When use atmega128 the function works, and return 29040, but atmega169 return "" (empty string)
[image]

First i use JSON link to add support of 128.  when i see 169 supported , i remove dir from C:\Users\ThinkPad\AppData\Local\Arduino15\packages\MegaCore
git-clone master branch into dir : D:\arduino-1.8.19\hardware\MegaCore  
yesterday i remove this dir of git clone,  then add board support from arduino hardware manager menu 2.2.0
I will test again when i go home

happysoul, 2022-06-14:
For read write single chip of game card (NES) ,China version called FC 
The full version will support 27Cxx 28xx 29xx with 28pin and 32pin (after test will add 12V support to write chip 27xx)
But the problem is atmega io output is weak, maybe drop this plan(write with many errors, although pull up every address and io with 1K to VCC), use atmega8a + 3 chip 74hc595 connect 29 chip address, 
Idea from https://github.com/walhi/arduino_eprom27_programmer (support 27cXX)
Copy from kazzo, made a fc card read-write programmer https://oshwhub.com/firseve/kazzo_smd 

Atmega169PA , arduino shows 86% ram cost (860 bytes), maybe i used more ram , then got this result, not enouth for local part
but ... other var works well , uint32_t saved 00000 and 7ffff, its convert from char* to number
169 1k ram,  128 4k ram

happysoul, 2022-06-14:
atmega169pa is cheaper then atmega8a, but has more io connect then 8a, 
Maybe 8a supported with arduino  [image]

MCUdude, 2022-06-14:
I suspect you're running out of RAM in your program. You can check the amount of free RAM at any time in your program using the function below. It will return the amount of free RAM in bytes.

```c
int freeRam () {
  extern int __heap_start, *__brkval; 
  int v; 
  return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval); 
}
```

> atmega169pa is cheaper then atmega8a, but has more io connect then 8a,
Maybe 8a supported with arduino

The Atmega8 can be used with Arduino IDE using [MiniCore](https://github.com/MCUdude/MiniCore). Note that it has even less RAM and flash memory than the ATmega169 has though.

happysoul, 2022-06-14:
Got it! Thank you
Get used to using java and python, ram is always enouth.
But hardware dev focus on every byte of ram ...

MCUdude, 2022-08-07:
I'm closing this issue as there's probably a RAM issue and not an issue with MegaCore.

closed 2022-08-07 by MCUdude