MCUdude/MegaCore #191: error while burning bootloader

jeremi99, 2022-07-04:
Hi. I probably have a common problem, but any help would be greatly appreciated.
I like to use the Arduino IDE (Currently version 1.8.16) for my Arduino Boards and I was happy to see that I could also load my programs with the IDE on Stand alone AVR chips, after loading MegaCore Boards with the Arduino Board Manager.
The difficulty I have is I cannot load the bootloader on my so called "Atmega128 minimum board".
The ISP programmer I am using is a USBasp bought on Aliexpress. My USBasp programmer will connect to m128 with AVRDUDE only if I add the option -B4, as follows:
AVRDUDE -p atmega128 -c usbasp -B4. 
It seems there is a SCK frequency problem if it is more than 187500 Hz.
Could that be the reason I cannot load the Bootloader onto my m128 ? Or else, could I load the bootloader with AVRDUDE directly ?
Thank you very much in advance for your valuable help.
Jeremi99

jeremi99, 2022-07-04:
Hi. Further to my previous comment, I found that the file "optiboot_atmega128.hex" was missing in the the (windows 10) Arduino directory "arduino/hardware/arduino/avr/bootloaders/optiboot".
Could that be a reason why the TOOLS>BURN BOOTLOADER option did not work and therefore "Error while burning bootloader" was shown ?
Thank you very much once more.

MCUdude, 2022-07-04:
Hi!

The default USBasp ISP clock is way too fast for a blank ATmega128 running at 1 MHz. If you want to bootload it through Arduino IDE, you can choose the _USBasp (slow) (MegaCore)_ option under the _Programmers_ menu. This option adds `-B32`, which is more than slow enough.

[image]

The Atmega128 bootloader file is supplied by this core, and can be found here:
https://github.com/MCUdude/MegaCore/tree/master/avr/bootloaders/optiboot_flash/bootloaders/atmega128

MCUdude, 2022-08-07:
OP didn't reply back. I'll assume the issue has been resolved.

closed 2022-08-07 by MCUdude