MCUdude/MegaCore #192: Arduino IDE can't access AVR ISP mk2

cerickson888, 2022-08-10:
I'm only able to access com ports but, as a true USB device, my programmer isn't assigned a port number. Can anyone suggest a way?

MCUdude, 2022-08-10:
The AVRISPmkII isn't a serial device, but native USB. It doesn't have a COM port. Simply select it in the programmers menu and hold shift while pressing upload (Upload using programmer).

cerickson888, 2022-08-10:
Thanks for the advice. I read that shift+upload would flash a sketch onto the target via a programmer (erasing the bootloader) but the bootloader is what I'm trying to flash the target with. I'd like to be able to use it like a normal Arduino board.

This post seems to show a way for Arduino IDE to use a USB port.
https://github.com/MCUdude/MegaCore/issues/172

     System wide configuration file is "C:\Users\syph3rd\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf"

     Using Port                    : usb
     Using Programmer              : atmelice_isp

But I don't know how to make that work.

MCUdude, 2022-08-10:
If you want to burn the bootloader, select the settings you prefer in the tools menu, and click "Burn Bootloader". If you're getting an error, please post it here.

cerickson888, 2022-08-10:
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ATmega1280, Yes (UART0), EEPROM retained, Arduino MEGA pinout, BOD 2.7V, LTO disabled, External 8 MHz"

avrdude: Version 6.3-20190619

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\A\Documents\Arduino\hardware\MegaCore-master\avr/avrdude.conf"

         Using Port                    : usb

         Using Programmer              : stk500v2

Error while burning bootloader.

avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

it says "         Using Programmer              : stk500v2" but the selected programmer is "AVRISP mkII (MegaCore)".

cerickson888, 2022-08-10:
avrdude: Version 6.3-20201216Failed chip erase: uploading error: exit status 1

I tried Arduino IDE version 2.0.0 rc9.2 cuz I'm out of ideas but that didn't work either

MCUdude, 2022-08-10:
This is most likely due to the AVRISPmkII using the incorrect driver. Instead of WinUSB, it has to be switched to libusb in order for it to work with Avrdude.

* Download Zadig: https://zadig.akeo.ie
* Select "List all devices" under the Options menu
* Look for AVRISPmkII in the drop down menu
* Switch driver to libusb or libusbk
* Retry with Avrdude

cerickson888, 2022-08-10:
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\A\Documents\Arduino\hardware\MegaCore-master\avr/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): Found AVRISP mkII, serno: 001D2C990079
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Firmware Version Master : 1.24
         Vtarget         : 4.8 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9702 (probably m128)
avrdude: erasing chip
avrdude: reading input file "0xff"
avrdude: writing lock (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xff:
avrdude: load data lock data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% -0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0x3f instead of 0xff (double check with your datasheet first).
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xff"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xff:
avrdude: load data efuse data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0b11000110"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0b11000110:
avrdude: load data hfuse data from input file 0b11000110:
avrdude: input file 0b11000110 contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% -0.00s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0b10111111"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0b10111111:
avrdude: load data lfuse data from input file 0b10111111:
avrdude: input file 0b10111111 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified

avrdude done.  Thank you.

 ***failed;  
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0x0f instead of 0xcf (double check with your datasheet first).

The driver seems to be working but still no joy.

cerickson888, 2022-08-11:
I really appreciate you taking the time to help me out with this. I suppose it's just something to do with the programmer I'm using so I'll buy another one and hope for the best..
[image]

MCUdude, 2022-08-11:
No worries!
It might be something with your programmer, I don't know.
The cheapest and most reliable AVR programmer is probably the USBasp. I'll recommend getting this one!

cerickson888, 2022-08-13:
I just received a new AVRISP mk2 (original not a copy) and it showed-up as a com port device right away allowing me to "Get Board Info" from Arduino IDE. However, it shows-up in Device Manager under "Other Devices" and does not have a driver installed. I searched for one online but couldn't find one. The AVRISP mk2 appeared in Zadig as "USB Serial" so I tried replacing the driver with libusbK but that just resulted in one less com port device in system. Any suggestions?

MCUdude, 2022-08-13:
The original AVRISPmkII is not a UAB to a serial device, so it should not show up as a COM port. I have an original one myself, so I can confirm. Are you 100% sure it is a genuine one?

If you use the libusbk driver, it should work with Avrdude if you have a working AVRISPmkII programmer.

cerickson888, 2022-08-14:
I realize this is totally not your problem so I'm going to stop bugging you about it. For what it's worth, I'm in Shenzhen where I buy almost everything from Taobao. You're supposed to be able to get authentic stuff from jd.com but, maybe not.

Not a hardcore embedded guy but I used PIC's back when they required a UV eraser, AVRs well before Arduino was popular, STM's cuz they're cheap enough for production.. never had this much trouble just trying to code a chip. Ironically, I wanted to run Arduino on a Atmega128 to make things easier i.e. even with 53 io's I still have to mux the inputs but at least I can drive 12 motors directly. Pluss most of the LSM6DS3 libraries are written for Arduino.     eh, some way some how.. :)

closed 2022-08-14 by cerickson888

mcuee, 2022-12-21:
Just FYI, there are many AVRISP mkii clones from Taobao and some of them are not working well at all.