MCUdude/MegaCore #194: ATmega 128 I2C doesn't work

krchan, 2022-11-07:
I have been using megacore for a standalone project on Atmega 128. I have followed the procedures. I was able use most of libraries  but when it comes to I2C, it is not working. Is there a configuration/setup change there. Using a pullup register also in the SDA/SCL pins. The I2C Scanner example code gets struck at Scanning.....

MCUdude, 2022-12-25:
It may be that your i2c target device is pulling the SDA or SCL line low. This will pretty much freeze the microcontroller until released because the current Wire library doesn't support timeouts yet. It's in the works though. Can you try with a different i2c device, just to check if there's anything wrong with your setup?

I have an ATmega128 myself, and it works perfectly fine with the Wire library

krchan, 2023-01-06:
I switched to another board and found the wire library to work perfectly
with the same devices.

Thank you for the support and help.

closed 2023-08-30 by MCUdude