MCUdude/MegaCore #197: How to wake up from sleep using WDT reset?

MFahreza69, 2023-01-31:


MFahreza69, 2023-01-31:
I am new in microcontroller programming, and i still confuse for waking up atmega64 from sleep using WDT reset because its not like atmega328 who has WDT interrupt, anyone can explain or give some simple code?
Thank You

MCUdude, 2023-01-31:
It seems like you can't wake up using the WDT on the ATmega64. It's an older device compared to newer chips like the ATmega1281, so its features are a bit limited.

From the ATmega64 datasheet:
[image]

From the ATmega328P datasheet:
[image]

MFahreza69, 2023-01-31:
Thank you for your response :D, i see that atmega64 hasn't wake up source using WDT
but is that possible for using the WDT timeout period and reseting the chip for wake up source?