MCUdude/MegaCore #198: pinMode does'nt work on ATmega128

GHOST-mHBr, 2023-04-14:
I have an ATmega128 micro-controller and i am trying to just program it!
I successfully program it with MegaCore boot loader (because the LED is blinking twice every second) using Arduino as isp programmer. then when i upload the following code on it (again with arduino as isp), the led does not blink
```C
void setup() {
  pinMode(15,OUTPUT);
}

void loop() {
  digitalWrite(15,1);
  delay(200);
  digitalWrite(15,0);
  delay(200);
}
```

but when i program the I/O registers myself, the led works correctly. the following code works:
```C
void setup() {
  DDRB = PORTB|(1<<PB5);
}

void loop() {
  PORTB |= (1<<PB5);
  delay(200);
  PORTB &= ~(1<<PB5);
  delay(200);
}
```
 i'm not sure where is the problem

also there is another problem with programming the micro-controller, after uploading the boot loader (MegaCore), i can't program it by USB to serial module. AVRdude's logs say that the response code is wrong:
```
         Using Port                    : /dev/ttyUSB2
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude ser_recv() OS error: unable to read: Resource temporarily unavailable
avrdude stk500_recv() error: programmer is not responding
avrdude stk500_getsync() warning: attempt 1 of 10: not in sync: resp=0x00
avrdude stk500_getsync() error: cannot communicate with device: resp=0xfc
avrdude main() error: unable to open programmer arduino on port /dev/ttyUSB2
```

GHOST-mHBr, 2023-04-14:
this is the schematic
[image]

MCUdude, 2023-08-30:
It is because you're using the wrong Arduino pin when trying to toggle the LED. You're using `15` instead of `13`. 15 is the physical pin of that particular package.

This will work:

```c
void setup() {
  pinMode(13,OUTPUT);
}

void loop() {
  digitalWrite(13,1);
  delay(200);
  digitalWrite(13,0);
  delay(200);
}
```

But this is what I'd prefer:

```c
void setup() {
  pinMode(PIN_PB5, OUTPUT);
}

void loop() {
  digitalWrite(PIN_PB5, 1);
  delay(200);
  digitalWrite(PIN_PB5, 0);
  delay(200);
}
```

closed 2023-08-30 by MCUdude