MCUdude/MegaCore #200: unable to burn bootloader for mega2560

BainesBunch, 2023-04-24:
avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : /dev/ttyS0
         Using Programmer              : stk500v2
avrdude ser_open() error: cannot open port /dev/ttyS0: The system cannot find the path specified.

avrdude main() error: unable to open programmer stk500v2 on port /dev/ttyS0

avrdude done.  Thank you.

Failed chip erase: uploading error: exit status 1

MCUdude, 2023-04-24:
The error message is pretty self-explanatory, isn't it? `/dev/ttyS0` isn't available.

BainesBunch, 2023-04-24:
yeah but it work fine with the standard mega2560 boards if i switch between the two, this only fails on the megacore

MCUdude, 2023-04-24:
Can you turn on verbose upload under the IDE settings, and paste the entire output here?

MCUdude, 2023-04-24:
The difference between the official Arduino boards and MegaCore in terms of uploading is that MegaCore uses Avrdude 7.1, while The standard Arduino boards use Avrdude 6.3

BainesBunch, 2023-04-24:
This is the standard mega flashing the bootloader
...........................................................................................................
"C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -patmega2560 -cstk500v2 -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): Found AVRISP mkII, serno: 000200216512
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Firmware Version Master : 1.24
         Vtarget         : 4.9 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: erasing chip
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
[... 112 lines cut ...]
avrdude: load data flash data from input file C:\Users\Peter\AppData\Local\arduino15\packages\arduino\hardware\avr\1.8.6/bootloaders/stk500v2/stk500boot_v2_mega2560.hex:
avrdude: input file C:\Users\Peter\AppData\Local\arduino15\packages\arduino\hardware\avr\1.8.6/bootloaders/stk500v2/stk500boot_v2_mega2560.hex contains 261406 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 261406 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

BainesBunch, 2023-04-24:
So why is 7 failing but 6 works OK

BainesBunch, 2023-04-24:
can we get megacore to point to 6 for avrdude ?

BainesBunch, 2023-04-24:
so it should look like this ?

avrispmkii.name=AVRISP mkII
avrispmkii.communication=usb
avrispmkii.protocol=stk500v2
avrispmkii.program.protocol=stk500v2
avrispmkii.program.tool=avrdude
avrispmkii.program.extra_params=-Pusb

BainesBunch, 2023-04-24:
progress but not just there thet, we now have the following errors
..........................................................................................................................................

"C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2/avrdude.conf" -v -patmega2560 -cstk500v2 -Pusb -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0b11010110:m -Ulfuse:w:0xf7:m

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude usbdev_open() error: cannot open device: Permission denied
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)
avrdude main() error: unable to open programmer stk500v2 on port usb

avrdude done.  Thank you.

Failed chip erase: uploading error: exit status 1

BainesBunch, 2023-04-24:
> What happens if you replace the entire avrispmkii section with this?
>
> (replaced stk500v2 with avrispmkii)
>
> avrispmkii.name=AVRISP mkII
> avrispmkii.communication=usb
> avrispmkii.protocol=avrispmkii
> avrispmkii.program.protocol=avrispmkii
> avrispmkii.program.tool=avrdude
> avrispmkii.program.extra_params=

Away from my pc for a couple of hours. My dog want walking and is not gonna take a pass LOL I’ll try this when I get back thanks a lot for your help

Cheers Pete

BainesBunch, 2023-04-24:
Nope unfortunatly that did not work either

MCUdude, 2023-04-24:
Try pasting the following command in Windows command prompt:

This first:
```
"C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2/avrdude.conf" -v -patmega2560 -csvrispmkii -Pusb
```

And then this:
```
"C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -patmega2560 -csvrispmkii -Pusb
```

And post the output of each command here. I'm trying to figure out why Avrdude 6.3 works while Avrdude 7.1 doesn't.

BainesBunch, 2023-04-24:
I have modified my avrdude.cfg and programmers.txt files, can you send my some virgin copies so we are both singing from the same hymn sheet PLZ

BainesBunch, 2023-04-24:
Dont wory i grabbed them from the repository

BainesBunch, 2023-04-24:
OK so here are the results

First command ......

C:\Windows\System32>C:\Users\Peter>"C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2/avrdude.conf" -v -patmega2560 -csvrispmkii -Pusb
'C:\Users\Peter' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>
C:\Windows\System32>avrdude: Version 7.1-arduino.1
'avrdude:' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>         Copyright the AVRDUDE authors;
'Copyright' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS
'see' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>
C:\Windows\System32>         System wide configuration file is C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf
'System' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>
C:\Windows\System32>
C:\Windows\System32>avrdude programmer_not_found() error: cannot find programmer id svrispmkii
'avrdude' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>
C:\Windows\System32>Valid programmers are:
'Valid' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  2232hio            = 2232hio based on FT2232H with buffer and LEDs [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:523]
'2232hio' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  4232h              = FT4232H based generic programmer [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:566]
'4232h' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  adafruit_gemma     = Adafruit Trinket Gemma bootloader disguised as USBtiny [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1269]
'adafruit_gemma' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  arduino            = Arduino for bootloader using STK500 v1 protocol [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:435]
'arduino' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  arduino-ft232r     = Arduino: FT232R connected to ISP [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1076]
'arduino-ft232r' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  arduino_gemma      = Arduino Gemma bootloader disguised as USBtiny [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1253]
'arduino_gemma' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  arduinoisp         = Arduino ISP Programmer         [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1283]
'arduinoisp' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  arduinoisporg      = Arduino ISP Programmer         [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1297]
'arduinoisporg' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  atmelice           = Atmel-ICE (ARM/AVR) in JTAG mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1838]
[... 438 lines cut ...]
C:\Windows\System32>  xplainedmini       = Atmel AVR XplainedMini in ISP mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1785]
'xplainedmini' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  xplainedmini_dw    = Atmel AVR XplainedMini in debugWIRE mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1798]
'xplainedmini_dw' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  xplainedmini_tpi   = Atmel AVR XplainedMini in TPI mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1825]
'xplainedmini_tpi' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  xplainedmini_updi  = Atmel AVR XplainedMini in UPDI mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1811]
'xplainedmini_updi' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  xplainedpro        = Atmel AVR XplainedPro in JTAG mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1744]
'xplainedpro' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  xplainedpro_pdi    = Atmel AVR XplainedPro in PDI mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1757]
'xplainedpro_pdi' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>  xplainedpro_updi   = Atmel AVR XplainedPro in UPDI mode [C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf:1771]

BainesBunch, 2023-04-24:
and this is the second ......

C:\Windows\System32>"C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -patmega2560 -csvrispmkii -Pusb

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

avrdude: Can't find programmer id "svrispmkii"

Valid programmers are:
  2232HIO          = FT2232H based generic programmer [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:413]
  4232h            = FT4232H based generic programmer [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:441]
  89isp            = Atmel at89isp cable            [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1465]
  abcmini          = ABCmini Board, aka Dick Smith HOTCHIP [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1364]
  alf              = Nightshade ALF-PgmAVR, http://nightshade.homeip.net/ [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1324]
  arduino          = Arduino                        [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:361]
  arduino-ft232r   = Arduino: FT232R connected to ISP [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:765]
  arduinoisp       = Arduino ISP Programmer         [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:893]
  arduinoisporg    = Arduino ISP Programmer         [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:902]
  atisp            = AT-ISP V1.1 programming cable for AVR-SDK1 from <http://micro-research.co.th/>  [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1420]
  atmelice         = Atmel-ICE (ARM/AVR) in JTAG mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1163]
  atmelice_dw      = Atmel-ICE (ARM/AVR) in debugWIRE mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1187]
  atmelice_isp     = Atmel-ICE (ARM/AVR) in ISP mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1195]
  atmelice_pdi     = Atmel-ICE (ARM/AVR) in PDI mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1171]
  atmelice_updi    = Atmel-ICE (ARM/AVR) in UPDI mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1179]
  avr109           = Atmel AppNote AVR109 Boot Loader [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:918]
  avr910           = Atmel Low Cost Serial Programmer [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:722]
  avr911           = Atmel AppNote AVR911 AVROSP    [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:925]
  avrftdi          = FT2232D based generic programmer [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:387]
  avrisp           = Atmel AVR ISP                  [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:609]
  avrisp2          = Atmel AVR ISP mkII             [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:629]
  avrispmkII       = Atmel AVR ISP mkII             [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:623]
  avrispv2         = Atmel AVR ISP V2               [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:616]
  bascom           = Bascom SAMPLE programming cable [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1318]
  blaster          = Altera ByteBlaster             [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1442]
  bsd              = Brian Dean's Programmer, http://www.bsdhome.com/avrdude/ [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1273]
  buspirate        = The Bus Pirate                 [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:634]
  buspirate_bb     = The Bus Pirate (bitbang interface, supports TPI) [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:641]
  butterfly        = Atmel Butterfly Development Board [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:911]
[... cut ...]
  xplainedmini_updi = Atmel AVR XplainedMini in UPDI mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1155]
  xplainedpro      = Atmel AVR XplainedPro in JTAG mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1123]
  xplainedpro_updi = Atmel AVR XplainedPro in UPDI mode [C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf:1131]

MCUdude, 2023-04-24:
OK, there is a typo in both commands, and that's my fault. `-csvrispmkii` should be `-cavrispmkii` instead.

> First command ......
C:\Windows\System32>C:\Users\Peter>"C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2/avrdude.conf" -v -patmega2560 -csvrispmkii -Pusb
'C:\Users\Peter' is not recognized as an internal or external command,
operable program or batch file.

You'll have to do your part to get the command to actually run. It looks like a copy/paste mistake to me.

BainesBunch, 2023-04-24:
First one .......
C:\Windows\System32>"C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2/avrdude.conf" -v -patmega2560 -cavrispmkii -Pusb

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf

         Using Port                    : usb
         Using Programmer              : avrispmkii
avrdude usbdev_open() error: cannot open device: Permission denied
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)
avrdude main() error: unable to open programmer avrispmkii on port usb

avrdude done.  Thank you.

Second one .........

C:\Windows\System32>"C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -patmega2560 -cavrispmkii -Pusb

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : avrispmkii
avrdude: usbdev_open(): Found AVRISP mkII, serno: 000200216512
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel AVR ISP mkII
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Firmware Version Master : 1.24
         Vtarget         : 4.9 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

BainesBunch, 2023-04-24:
Sooooooo Second one works ~~~~~

MCUdude, 2023-04-24:
It's most likely a driver issue. I'm not sure why. Avrdude 7.1 should provide better compatibility, not worse. @mcuee do you have any idea what going on here?

Here's what you can do:

* Download Zadig: http://zadig.akeo.ie/.
* Plug in your AVRISPmkII
* Start Zadig
* From the dropdown menu, select AVRISPmkII. If it doesn't show up on the menu, then select Options > List all devices and check again.
* Which driver is currently loaded for the AVRISPmkII?
* Try switching to WinUSB and try the first Avrdude command (Avrdude 7.1) again.

BainesBunch, 2023-04-24:
OK with the WinUSB driver loaded here is the first command result

C:\Windows\System32>"C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -patmega2560 -cavrispmkii -Pusb

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : avrispmkii
avrdude: usbdev_open(): Found AVRISP mkII, serno: 000200216512
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel AVR ISP mkII
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Firmware Version Master : 1.24
         Vtarget         : 4.9 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

C:\Windows\System32>"C:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\MegaCore\hardware\avr\2.2.2/avrdude.conf" -v -patmega2560 -cavrispmkii -Pusb

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
[... 41 lines cut ...]
         Vtarget         : 4.9 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)

avrdude done.  Thank you.

and here is the second command 

C:\Windows\System32>"C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -patmega2560 -cavrispmkii -Pusb

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Peter\AppData\Local\arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : avrispmkii
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

C:\Windows\System32>

MCUdude, 2023-04-24:
That seems about right. WinUSB will only work with Avrdude 7.1.
So with WinUSB selected, you can use your AVRISPmkII with MegaCore.

But which USB driver was previously selected that did work with Avrdude 6.3 but not with Avrdude 7.1?

BainesBunch, 2023-04-24:
So the WinUSB (v6.1.7600.16385) drivers work with the avrdude: Version 7.1-arduino.1 but i need to use the libusb-win32 (v1.2.7.3) drivers with avrdude: Version 6.3-20190619

MCUdude, 2023-04-24:
The thing is that Avrdude 7.1 should work both libusb and WinUSB. I'm not sure why libusb-win32 doesn't work for you.

As a last test, can you try using libusbK, and see if this works with both 6.3 and 7.1?

BainesBunch, 2023-04-24:
It does not they seem to be mutualy exclusive :(

BainesBunch, 2023-04-24:
Now that i have it compiling using the arduino IDE my next trick it to get it working through platformio

do you have a platformIO files that you know works with this combination of drivers and MCU ?

MCUdude, 2023-04-24:
> It does not they seem to be mutualy exclusive :(

So, libusbK does work with both, or none?

But on Windows, WinUSB is the preferred driver, since Microchip Studio 7 or MPLAB X will work with the WinUSB driver, and not libusb.

> do you have a platformIO files that you know works with this combination of drivers and MCU ?

Yes, See https://github.com/MCUdude/MegaCore/blob/master/PlatformIO.md for details

This should, however, suit your needs, given that your ATmega2560 is running at 16 MHz.

```ini
; PlatformIO template configuration file for MegaCore
; https://github.com/MCUdude/MegaCore/
;
;   Build options: build flags, source filter
;   Hardware options: oscillator type, BOD, UART number, EEPROM retain
;   Upload options: custom upload port, speed, and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options
; https://github.com/MCUdude/MegaCore/blob/master/PlatformIO.md
; https://docs.platformio.org/page/projectconf.html
; https://docs.platformio.org/en/latest/platforms/atmelavr.html

[platformio]
default_envs = Upload_UART ; Default build target

; Common settings for all environments
[env]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; Chip in use
board = ATmega2560
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L

; BUILD OPTIONS
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
build_flags = 

; SERIAL MONITOR OPTIONS
; Serial monitor port defined in the Upload_UART environment
monitor_port = ${env:Upload_UART.upload_port}
; Serial monitor baud rate
monitor_speed = 9600

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
;  Serial bootloader protocol
upload_protocol = arduino
; Serial upload port
upload_port =
; Get upload baud rate defined in the fuses_bootloader environment
board_upload.speed = ${env:fuses_bootloader.board_bootloader.speed}

; Run the following command to upload with this environment
; pio run -e Upload_ISP -t upload
[env:Upload_ISP]
; Custom upload procedure
upload_protocol = custom
; Avrdude upload flags
upload_flags =
  -C$PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
  -p$BOARD_MCU
  -PUSB
  -cavrispmkii
; Avrdude upload command
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i

; Run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; Run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader
[env:fuses_bootloader]
board_hardware.oscillator = external ; Oscillator type
board_hardware.uart = uart0          ; Set UART to use for serial upload
board_bootloader.speed = 115200      ; Set bootloader baud rate
board_hardware.bod = 2.7v            ; Set brown-out detection
board_hardware.eesave = yes          ; Preserve EEPROM when uploading using programmer
upload_protocol = avrispmkii         ; Use the USBasp as programmer
upload_flags =                       ; Select USB as upload port and divide the SPI clock by 8
  -PUSB
  -B8
```

Use
`pio run -t bootloader -e fuses_bootloader` to burn the bootloader though PlatformIO using your AVRISPmkII`

Use
`pio run -t upload -e Upload_ISP` to upload using your AVRISPmkii

Use
`pio run -t upload` to upload using the bootloader. You don't need to specify an environment (-e) here, because Upload_UART is set as the standard environment at the top on the ini file.

BainesBunch, 2023-04-24:
thanks so much for your help, please drop a note back here if you ever figure out why those drivers are not behaving as expected.

mcuee, 2023-04-24:
> thanks so much for your help, please drop a note back here if you ever figure out why those drivers are not behaving as expected.

As mentioned by Hans, the preferred driver for AVRISP mkII is WinUSB, since Microchip Studio 7 or MPLAB X will work with the WinUSB driver, and not libusb0.sys.

If you really want to use both Arduino's version of avrdude 6.3 (using libusb-win32 library) and 7.1 (using libusb-1.0 + libusb-compat-0.1 library), then you have to use libusbK driver.

mcuee, 2023-04-25:
I do not have the ATmega2560 with me right now, so I tested with ATmega328P and I can not reproduce the following issue using MiniCore (with https://github.com/MCUdude/MiniCore/commit/6d7701976bf2a5008570843b9cc144aa59515638), all three drivers work fine.

Run log with libusb0.sys 1.2.6.0 driver.

```
C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/avrdude.conf -v -patmega328p -cavrispmkii -e -Ulock:w:0xff:m -Uefuse:w:0b11111101:m -Uhfuse:w:0b11010110:m -Ulfuse:w:0b11110111:m 

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : usb
         Using Programmer              : avrispmkii
avrdude: usbdev_open(): found AVRISP mkII, serno: 001D2C990079
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel AVR ISP mkII
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Serial number   : 001D2C990079
         Firmware Version Controller : 1.24
         Vtarget         : 4.8 V
         SCK period      : 4.00 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: reading input file 0xff for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xff
avrdude: 1 byte of lock verified
avrdude: reading input file 0b11111101 for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0b11111101
avrdude: 1 byte of efuse verified
C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/avrdude.conf -v -patmega328p -cavrispmkii -Uflash:w:C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex:i -Ulock:w:0xcf:m 
avrdude: reading input file 0b11010110 for hfuse
[... 63 lines cut ...]
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex for flash
         with 484 bytes in 2 sections within [0x7e00, 0x7fff]
         using 4 pages and 28 pad bytes
avrdude: writing 484 bytes flash ...

Writing | ################################################## | 100% 0.02s

avrdude: 484 bytes of flash written
avrdude: verifying flash memory against C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex

Reading | ################################################## | 100% 0.00s

avrdude: 484 bytes of flash verified
avrdude: reading input file 0xcf for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xcf
avrdude: 1 byte of lock verified

avrdude done.  Thank you.

```

mcuee, 2023-04-25:
@BainesBunch 
Basically I can not reproduce the issue. Just wondering how you install libusb0.sys driver? You should use Zadig. Latest Zadig 2.8 version will install libusb0.sys 1.2.7.3 version. Older Zadig 2.7 version will install libusb0.sys 1.2.6.0 version.

```
avrdude usbdev_open() error: cannot open device: Permission denied
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)
avrdude main() error: unable to open programmer avrispmkii on port usb
```

MCUdude, 2023-04-25:
@mcuee I suspect he used the libusb driver that was bundled with Arduino IDE (libusb0.dll, found in Arduino/hardware/tools/avr/bin). My (work) computer says this is version 1.2.6.0:

[image]

mcuee, 2023-04-25:
No issues with ATmega2560 either with MegaCore (with https://github.com/MCUdude/MegaCore/commit/12e390c62825ff7e3fe328ffbe17756498e465d4 fix).

```
C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.2/avrdude.conf -v -patmega2560 -cavrispmkii -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0b11010110:m -Ulfuse:w:0xf7:m 

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.2\avrdude.conf

         Using Port                    : usb
         Using Programmer              : avrispmkii
avrdude: usbdev_open(): found AVRISP mkII, serno: 001D2C990079
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel AVR ISP mkII
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Serial number   : 001D2C990079
         Firmware Version Controller : 1.24
         Vtarget         : 4.9 V
         SCK period      : 4.00 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude: erasing chip
avrdude: reading input file 0xff for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xff
avrdude: 1 byte of lock verified
avrdude: reading input file 0xfd for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0xfd
avrdude: 1 byte of efuse verified
avrdude: reading input file 0b11010110 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
[... 61 lines cut ...]
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.2/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex for flash
         with 922 bytes in 2 sections within [0x3fc00, 0x3ffff]
         using 4 pages and 102 pad bytes
avrdude: writing 922 bytes flash ...

Writing | ################################################## | 100% 0.05s

avrdude: 922 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.2/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex

Reading | ################################################## | 100% 0.00s

avrdude: 922 bytes of flash verified
avrdude: reading input file 0xcf for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xcf
avrdude: 1 byte of lock verified

avrdude done.  Thank you.
```

mcuee, 2023-04-25:
@BainesBunch 
Make sure you use Zadig to install libusb-win32 device driver for your AVRISP mkii. In order for avrdude 7.1 to work with libusb0.sys or libusbK.sys, libusbK.dll is required. If you install using Zadig, it will automatically install libusbK.dll for you.

mcuee, 2023-04-25:
@BainesBunch 

Please also read libusb Wiki here. 
https://github.com/libusb/libusb/wiki/Windows#user-content-Driver_Installation
https://github.com/libusb/libusb/wiki/FAQ#libusb-win32-libusbk-and-libusb-project-what-about-openusb 

Yes I agree it is rather complicated but that is the reality to use libusb under Windows.

FYI: I am the non-developer admin of libusb, libusb-win32 and libusbK project, as well as avrdude project, mainly on testing and supporting side, as I can not code.

mcuee, 2023-05-19:
I think the issue can be closed.