MCUdude/MegaCore #201: stk500v2 bootloader potential issues

mcuee, 2023-06-10:
Reference:
* https://github.com/avrdudes/avrdude/issues/1379#issuecomment-1585323425

Other than the EEPROM read issue fixed in this repo, there are also other issues reported in Arduino.
https://github.com/arduino/ArduinoCore-avr/issues?q=is%3Aissue+is%3Aopen+stk500v2

And there are other isues reported here.
https://github.com/msproul/Arduino-stk500v2-bootloader/issues

mcuee, 2023-06-10:
@MCUdude 

* https://github.com/arduino/ArduinoCore-avr/issues/173 (I believe still exist in this repo)
* https://github.com/arduino/ArduinoCore-avr/issues/172 (I believe still exist in this repo)
* https://github.com/arduino/ArduinoCore-avr/issues/191 (still exist in this repo, confirmed)
* https://github.com/arduino/ArduinoCore-avr/issues/201  (I believe still exist in this repo)
* https://github.com/msproul/Arduino-stk500v2-bootloader/issues/7 ( I believe still exist in this repo)

I have not checked the details of the other issues reported here.
https://github.com/msproul/Arduino-stk500v2-bootloader/issues

mcuee, 2023-06-10:
* https://github.com/arduino/ArduinoCore-avr/issues/201

The following mod will still run (LED is blinking)
https://github.com/arduino/Arduino-stk500v2-bootloader/blob/master/test/bigprogram/bigprogram.ino

But pgm_read_byte does not wotk due to the following issue.
* https://github.com/arduino/ArduinoCore-avr/issues/173

```
const char check[] PROGMEM = "A nice string!\n";

void setup() {                
  // initialize the digital pin as an output.
  // Pin 13 has an LED connected on most Arduino boards:
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
}

void SendSTR_P(const char *pFlashSTR) {
  uint8_t i;  
  for (i = 0; pgm_read_byte(&pFlashSTR[i]); i++)
    Serial.print(pgm_read_byte(&pFlashSTR[i]));
  ;
}

void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // set the LED off
  delay(1000);              // wait for a second
  digitalWrite(LED_BUILTIN, HIGH);    // set the LED on
  delay(1000);              // wait for a second
  if (millis()<1000) {
    SendSTR_P(aa);
    SendSTR_P(bb);
//    SendSTR_P(cc);
//    SendSTR_P(dd);
//    SendSTR_P(ee);
//    SendSTR_P(ff);
//    SendSTR_P(gg);
 //   SendSTR_P(hh);
 //   SendSTR_P(ii);
  }
  SendSTR_P(check);
}
```

mcuee, 2023-06-10:
This mod will make the LED blinking no longer works, even though `SendSTR_P(check)` still works.

```
const char check[] PROGMEM = "A nice string!\n";

void setup() {                
  // initialize the digital pin as an output.
  // Pin 13 has an LED connected on most Arduino boards:
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
}

void SendSTR_P(const char *pFlashSTR) {
  uint8_t i;  
  for (i = 0; pgm_read_byte(&pFlashSTR[i]); i++)
    Serial.print(pgm_read_byte(&pFlashSTR[i]));
  ;
}

void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // set the LED off
  delay(1000);              // wait for a second
  digitalWrite(LED_BUILTIN, HIGH);    // set the LED on
  delay(1000);              // wait for a second
  if (millis()<1000) {
    SendSTR_P(aa);
    SendSTR_P(bb);
    SendSTR_P(cc);
//    SendSTR_P(dd);
//    SendSTR_P(ee);
//    SendSTR_P(ff);
//    SendSTR_P(gg);
 //   SendSTR_P(hh);
 //   SendSTR_P(ii);
  }
  SendSTR_P(check);
}

Sketch uses 98542 bytes (38%) of program storage space. Maximum is 253952 bytes.
Global variables use 184 bytes (2%) of dynamic memory, leaving 8008 bytes for local variables. Maximum is 8192 bytes.
```

MCUdude, 2023-11-30:
I decided to abandon the stk500v2 bootloader and instead go all in with Urboot.

closed 2023-11-30 by MCUdude