MCUdude/MegaCore #203: Is it possible to program with FT232 programming? Is there a schematic drawing available?

uguraltinsoy, 2023-07-20:
Is it possible to program with FT232 programming? Is there a schematic drawing available?

MCUdude, 2023-07-23:
Yes, you can use any USB to serial chip to upload programs as long as a bootloader is present. This needs to be flashed with an ISP programmer, such as the USBtinyISP or USBasp.

> Is there a schematic drawing available?

Yes, here: https://github.com/MCUdude/MegaCore/tree/master#minimal-setup

uguraltinsoy, 2023-07-23:
can i use it without bootloader?

http://electronics-diy.com/FT232RL_USB_to_Serial_Adapter_for_PIC_AVR_ATMEGA_ARDUINO_MCUs.php

MCUdude, 2023-07-23:
> can i use it without bootloader?

Not with MegaCore. You need an ISP programmer to load the bootloader. Then you can use the F232RL to upload new programs.

uguraltinsoy, 2023-07-25:
Is there a resource on how to load the bootloader, can you guide me?

MCUdude, 2023-07-25:
There are plenty of resources online. YOu can start by reading the [MegaCore readme file](https://github.com/MCUdude/MegaCore/blob/master/README.md). All you need to do is to connect an ISP programmer to the appropriate pins, select your preferred settings in the IDE Tools menu, and hit "Burn bootloader".