MCUdude/MegaCore #204: Customizable LED_BUILTIN

marshfolx, 2023-08-13:
I build a mega128 board and want to put LED_BUILTIN on D16 / PG3 rather than D13 / PB5 which is a PWM channel of Timer1. I think it is better to put a #ifndef around the #define LED_BUILTIN in pins_arduino.h, so that I can use build flag to define my assignment.

```cpp
// Builtin LED
#ifndef LED_BUILTIN
#define LED_BUILTIN   (13)
#endif
static const uint8_t LED = LED_BUILTIN;
```

MCUdude, 2023-08-13:
`LED_BUILTIN` is just a constant, and isn't "tied" to a specific pin at all. You don't have to use the constant if you don't want to. However, you should be able to deal with it in your code like so:

```c
#ifdef LED_BUILTIN
#undef LED_BUILTIN
#define LED_BUILTIN PIN_PG3
#endif
```

marshfolx, 2023-08-13:
> `LED_BUILTIN` is just a constant, and isn't "tied" to a specific pin at all. You don't have to use the constant if you don't want to. However, you should be able to deal with it in your code like so:
> 
> ```c
> #ifdef LED_BUILTIN
> #undef LED_BUILTIN
> #define LED_BUILTIN PIN_PG3
> #endif
> ```

Some libraries use LED_BUILTIN to show simple notifications, I have to diretly change the source code if i can't use build flag. Or I must place `#undef LED_BUILTIN` before including the library.

MCUdude, 2023-08-13:
You can't override pre-defined macros in the sketch, and expect the compiler to pick it up and re-compile the libraries.

This means that the only viable way to override the LED_BUILTIN macro would either to add `-DLED_BUILTIN=13` as a compiler build flag (difficult with Arduino IDE, possible with PlatformIO), or modify the source code in the `MCUdude_corefiles` folder. I'm happy to add your suggestion to the code base (only defined LED_BUILTIN if not already defined), but it won't solve your issue. The compiler would still not pick this up as long as the macro is overridden in the sketch.

marshfolx, 2023-08-13:
Thanks for that, and of course I'm using PlatformIO. I realized that `#undef` can't affect library code just after I sent the comment. No problem about it.

closed 2023-08-13 by marshfolx