MCUdude/MegaCore #205: Arduino Mega OTEthernet not working

JaspervanKol, 2023-09-20:
I succesfully installed the MegaCore v2.2.3 in the Arduino IDE and burnt the bootloader to my Arduino Mega. I can upload any sketch over USB to the Arduino Mega but when I try to use the OTA lib it doesn't work.

Inside of the IDE I can select the Arduino Mega with the port for OTA as shown below:

[image]

But when I want to upload the sketch I'm getting this error:

```
avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\jaspe\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3\avrdude.conf

         Using Port                    : 192.168.2.237
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude ser_open() error: cannot open port 192.168.2.237: Het systeem kan het opgegeven bestand niet vinden.

avrdude main() error: unable to open programmer arduino on port 192.168.2.237

avrdude done.  Thank you.

Failed uploading: uploading error: exit status 1
```

Does somebody knows what I'm doing wrong? I really like to have this functionality to upload sketches over the internet...

MCUdude, 2023-09-20:
I've not been involved in the ArduinoOTA side of things. @JAndrassy knows more about this. But it looks to me like you're trying to use the ``arduino`` bootloader protocol to do OTA uploads using Avrdude instead of ArduinoOTA. This can't be right...

JAndrassy, 2023-09-20:
Did you copy extras/avr/platform.local.txt next to platform.txt as required?

JaspervanKol, 2023-09-20:
@JAndrassy no I did not. I followed the "**How to install**" & "**Getting started with MegaCore**" but I didn't see anything with the platform files.

Where can I find this file and where do I need to place this?

JAndrassy, 2023-09-20:
https://github.com/JAndrassy/ArduinoOTA#atmega-support

JaspervanKol, 2023-09-20:
Ok, I didn't know that. I did read the "**ATmega support**" description, but I still don't understand it. Where can I find this file and where do I need to place it on my pc?

JaspervanKol, 2023-09-22:
I figured it out! I needed to add the myboards from https://github.com/jandrassy/my_boards to the IDE. I can now upload new sketches over the internet using the new board, still thanks for the help. Closing...

closed 2023-09-22 by JaspervanKol