MCUdude/MegaCore #206: generation of the file .hex

Dancofra, 2023-09-20:
Excellent work, I think the generation of the file is not working. hex, all this is done from the Arduino id and throws the following error:

Dancofra, 2023-09-20:
Sketch uses 1646 bytes (0%) of program storage space. Maximum is 253952 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 8183 bytes for local variables. Maximum is 8192 bytes.
Acceso denegado.
exit status 1

Compilation error: exit status 1

MCUdude, 2023-09-20:
Open the IDE settings turn on Verbose compilation and post the entire output along with the code you're trying to compile.

There is also some information here:
https://support.arduino.cc/hc/en-us/articles/6554914611228-Compilation-error-exit-status-1

Dancofra, 2023-09-20:
FQBN: MegaCore:avr:2560:BOD=disabled,LTO=Os_flto,bootloader=original_mega_2560
Using board '2560' from platform in folder: C:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3

Detecting libraries used...
C:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g -IC:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3\cores\MCUdude_corefiles -IC:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3\variants\100-pin-arduino-mega C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\sketches\D69421F1BD6C3FAADDBCBE5A16D05ACF\sketch\Blink.ino.cpp -o nul
Generating function prototypes...
C:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g -IC:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3\cores\MCUdude_corefiles -IC:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3\variants\100-pin-arduino-mega C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\sketches\D69421F1BD6C3FAADDBCBE5A16D05ACF\sketch\Blink.ino.cpp -o C:\Users\DanierCondeFranco\AppData\Local\Temp\2320535586\sketch_merged.cpp
C:\Users\DanierCondeFranco\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\DanierCondeFranco\AppData\Local\Temp\2320535586\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.2.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.2.3\\variants\\100-pin-arduino-mega" "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF\\sketch\\Blink.ino.cpp" -o "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF\\sketch\\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core: C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\cores\d5bf60545c1bf7782c33b56b346fcc6e\core.a
Linking everything together...
"C:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -Wall -Os -Wl,--gc-sections,--section-start=.FAR_MEM1=0x10000,--section-start=.FAR_MEM2=0x20000,--section-start=.FAR_MEM3=0x30000 -mmcu=atmega2560 -w -flto -g -o "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.elf" "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF\\sketch\\Blink.ino.cpp.o" "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/..\\..\\cores\\d5bf60545c1bf7782c33b56b346fcc6e\\core.a" "-LC:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF" -lm
"C:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.elf" "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.eep"
"C:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.elf" "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.hex"
cmd /C echo. && "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.elf" > "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino_atmega2560_16000000L.lst"
 

[... cut ...]
"C:\\Users\\DanierCondeFranco\\AppData\\Local\\Arduino15\\packages\\MegaCore\\hardware\\avr\\2.2.3/scripts/delete_merged_output.bat" true "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF\\Blink.ino.with_bootloader.hex"
Acceso denegado.
cmd /C copy "C:\\Users\\DanierCondeFranco\\AppData\\Local\\Temp\\arduino\\sketches\\D69421F1BD6C3FAADDBCBE5A16D05ACF\\Blink.ino_atmega2560_16000000L.lst" {sketch_path}
        0 archivo(s) copiado(s).
exit status 1

Compilation error: exit status 1

MCUdude, 2023-09-20:
> Acceso denegado.

It seems like the IDE doesn't have access to execute one or more scripts. What happens if you try to run the IDE as administrator in Windows? @per1234 may know what the root cause is.

Dancofra, 2023-09-20:
I can load the blink example without problem using the Mega-core - atmega2560 . but when generating the .hex file “Export Compiled Binary” the problem is generated.

MCUdude, 2023-09-20:
Again, Arduino IDE doesn't have permission to run the scripts associated with exporting the compiled output and moving it into the sketch folder. I can't tell why this is; probably a limitation Windows puts on your account.

However, if you just want to get the hex file, you can quite easily look at the compiler output and figure out where the hex file is located.

```
C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\sketches\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.hex
```

per1234, 2023-09-20:
I'm wondering if the problem is the unexpanded `{sketch_path}` property reference in this command:

>```text
>"C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\sketches\D69421F1BD6C3FAADDBCBE5A16D05ACF\Blink.ino_atmega2560_16000000L.lst" {sketch_path}
>```

I see that also in the output from running an "Export Compiled Binary" operation with a MegaCore board selected. In my case it doesn't produce an error. What it does do is copy the `.lst` file to a file with the literal name `{sketch_path}` in the root of the Arduino IDE installation folder.

Maybe the reason why that generates an error for @Dancofra is that the folder where the command attempts to write this file has some security restrictions.

---

I guess there was a regression at some point in the platform system that causes the `sketch_path` property to no longer be available as promised in the specification:

https://arduino.github.io/arduino-cli/latest/platform-specification/#recipes-to-export-compiled-binary

Dancofra, 2023-09-20:
Thanks for your prompt responses. I will take the .hex file from the folder “C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\sketches\D69421F1BD6C3FAADDBCBE5A16D05ACF”

> Nuevamente, Arduino IDE no tiene permiso para ejecutar los scripts asociados con la exportación de la salida compilada y su movimiento a la carpeta de bocetos. No puedo decir por qué es esto; Probablemente sea una limitación que Windows pone en su cuenta.
> 
> Sin embargo, si sólo desea obtener el archivo hexadecimal, puede mirar fácilmente la salida del compilador y averiguar dónde se encuentra el archivo hexadecimal.
> 
> ```
> C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\sketches\D69421F1BD6C3FAADDBCBE5A16D05ACF/Blink.ino.hex
> ```

Thanks for your prompt responses. I will take the .hex file from the folder “C:\Users\DanierCondeFranco\AppData\Local\Temp\arduino\sketches\D69421F1BD6C3FAADDBCBE5A16D05ACF”

per1234, 2023-09-26:
>there was a regression at some point in the platform system that causes the sketch_path property to no longer be available as promised in the specification

I have now submitted a formal report about this bug:

https://github.com/arduino/arduino-cli/issues/2340

On investigation, I found it was not as straightforward as a simple regression.

I found that the support for the `savehex` [hooks](https://arduino.github.io/arduino-cli/latest/platform-specification/#pre-and-post-build-hooks-since-arduino-ide-165) that execute the problematic command was only recently added to Arduino CLI (https://github.com/arduino/arduino-cli/pull/2091). Previously this command (which is not actually essential but makes the assembly listing easier for the user to find) was never executed at all for Arduino CLI/Arduino IDE 2.x users. 

Since the addition of the support for the hook in Arduino CLI 0.32.0/Arduino IDE 2.1.0, the command is executed but Arduino CLI still doesn't generate the `sketch_path` platform property as promised in the specification so we end up with this behavior of the command running, but not being valid due to the unexpanded reference for the missing property.

mcuee, 2023-10-20:
https://github.com/arduino/arduino-cli/issues/2340 has been closed by PR https://github.com/arduino/arduino-cli/pull/2372.