MCUdude/MegaCore #207: cannot be compiled when using the Wire library

vanduy10061991, 2023-10-05:
D:\ideV8\portable\sketchbook\libraries\Wire\Wire.cpp: In member function 'void TwoWire::begin()':
D:\ideV8\portable\sketchbook\libraries\Wire\Wire.cpp:62:3: error: 'DIDR0' was not declared in this scope
   DIDR0 &= 0xCF;
   ^~~~~
D:\ideV8\portable\sketchbook\libraries\Wire\Wire.cpp:62:3: note: suggested alternative: 'OCDR0'
   DIDR0 &= 0xCF;
   ^~~~~
   OCDR0
Multiple libraries were found for "Wire.h"
 Used: D:\ideV8\portable\sketchbook\libraries\Wire
 Not used: D:\ideV8\portable\packages\MegaCore\hardware\avr\2.2.3\libraries\Wire
exit status 1
Error compiling for board ATmega128.

MCUdude, 2023-10-05:
Have a look at the error message. The IDE is using the wrong Wire library

vanduy10061991, 2023-10-05:
How can I select the Megacore hardware library when compiling it?  It does not choose the hardware library itself but runs the default library

MCUdude, 2023-10-05:
Remove the library it uses instead. Then it will switch back to the one provided by MegaCoee

vanduy10061991, 2023-10-06:
I tried that and it worked, but when I compiled it for another MCU, it gave me an error that it couldn't compile

MCUdude, 2023-10-19:
Well, that's not MegaCore's problem...

If you have two conflicting libraries, you'll have to deal with it some how. I suggest renaming the modified one.

closed 2023-10-19 by MCUdude