MCUdude/MegaCore #208: Cannot upload sketch to mega without bootloader using arduinoISP

fdasbfg, 2023-12-15:
Hello

I cannot have D13 blink at all during boot, it is attached to a low side driver which controls a fuel injector. I also have no use of a bootloader. Rather than mess with modifying a bootloader to not use D13, I would like to just upload a sketch using an uno as arduinoISP. 

I know the wiring is correct because I can burn a bootloader. However, if I try and use CTRL+SHIFT+U to upload using programmer megacore appears to try and use AVRDude...

"C:\\Users\\\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "C:\\Users\\\\AppData\\Local\\Temp\\arduino_build_726309/speeduino.ino.elf"
Sketch uses 178776 bytes (68%) of program storage space. Maximum is 262144 bytes.
Global variables use 7803 bytes (95%) of dynamic memory, leaving 389 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur.
C:\Users\\AppData\Local\Arduino15\packages\MegaCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3/avrdude.conf -v -patmega2560 -cstk500v1 -PCOM29 -b19200 -Uflash:w:C:\Users\\AppData\Local\Temp\arduino_build_726309/speeduino.ino.with_bootloader.hex:i 

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.3\avrdude.conf

         Using Port                    : COM29
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude ioerror() OS error: file C:\Users\\AppData\Local\Temp\arduino_build_726309/speeduino.ino.with_bootloader.hex is not readable: No such file or directory

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected
 
 Of course it cannot find a file that does not exist. 
 
Is there a quick fix for this?

Thank you!

MCUdude, 2023-12-15:
First, I'll recommend you to install the latest MegaCore v3.0.0 version.

MegaCore v3.0.0 uses a different bootloader than the previous versions, and it behaves a little differently.
The "bootloader LED" does not flash when you reset the board, only when you are actually uploading a new program.
And if you use PlatformIO, you can specify that you don't want any "bootloader LED".

If you want to upload without using a bootloader, select "No bootloader" in the tools menu, and click upload. No need to hold shift.

fdasbfg, 2023-12-15:
I think something is wrong with the core, then. I started with 3.0, but when I used a bootloader my custom PCB prevents the mega from booting. I've now managed to upload without the bootloader with your help using 3.0, and it still won't boot when it's installed on my custom PCB. 

If I use the default arduino core and bootloader it works just fine besides the two blinks on D13 at every boot. I've since figured out  that I can upload without a bootloader using the default core and I get no D13 blinks and everything works like it should when installed on my board. 

The only things I can gather that might cause issue, the code I am using in combination with the PCB pulls the reset pin high using a pin to prevent resets on serial connections. I might plug USB in while the code is running and I don't want the board to reset like I think they usually do till the auto-reset circuit charged up.  

I use D3, and my board has a 10K pullup and 0.1u capacitor on D3 and the reset pin so it should still be OK on a board not running code, unless this core drives D3 low? Aren't they usually left floating for power saving?

tomaskovacik, 2023-12-15:
If you mess with auto-reset setup try manual reset when uploading, it is the same as auto-reset also try to investigate with minimal/known good configuration = assemble your board in the same setup as Arduino which works, then add components, or try removing components from your custom board.

fdasbfg, 2023-12-15:
It works just fine with the default arduino core, either with or without the stk500 bootloader, so I have removed megacore since that is a lot easier than removing SMD components that obviously work fine. I don't have any problem uploading code, unless you try CTRL+SHIFT+U/upload using programmer. 

Still think you should be able to CTRL+SHIFT+U and have it function properly, as megacore is now it does not work. I'm no pro, but it doesn't try and upload the file it just built like it should.

MCUdude, 2023-12-15:
It may be because you haven't set the correct fuses.

Pick your options in the Tools menu and click "Burn bootloader", even though you aren't using a bootloader. This is the only way to set the fuses, and this label can't be changed by 3rd part cores.

I can't reproduce your issue with the correct fuses at least. The blink sketch are running just fine on my test board.

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/hans/Library/Arduino15/packages -hardware /Users/hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/hans/Documents/Arduino/libraries -fqbn=MegaCore:avr:649:bootloader=no_bootloader,eeprom=keep,variant=model649,BOD=2v7,LTO=Os_flto,clock=16MHz_external -ide-version=10815 -build-path /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_425295 -warnings=all -build-cache /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_cache_103448 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/Users/hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/Users/hans/Library/Arduino15/packages/MegaCore/tools/avrdude/7.2-arduino.1 -prefs=runtime.tools.avrdude-7.2-arduino.1.path=/Users/hans/Library/Arduino15/packages/MegaCore/tools/avrdude/7.2-arduino.1 -verbose /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_modified_sketch_753246/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/hans/Library/Arduino15/packages -hardware /Users/hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/hans/Documents/Arduino/libraries -fqbn=MegaCore:avr:649:bootloader=no_bootloader,eeprom=keep,variant=model649,BOD=2v7,LTO=Os_flto,clock=16MHz_external -ide-version=10815 -build-path /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_425295 -warnings=all -build-cache /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_cache_103448 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/Users/hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/Users/hans/Library/Arduino15/packages/MegaCore/tools/avrdude/7.2-arduino.1 -prefs=runtime.tools.avrdude-7.2-arduino.1.path=/Users/hans/Library/Arduino15/packages/MegaCore/tools/avrdude/7.2-arduino.1 -verbose /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_modified_sketch_753246/Blink.ino
Using board '649' from platform in folder: /Users/hans/Library/Arduino15/packages/MegaCore/hardware/avr/3.0.0
Using core 'MCUdude_corefiles' from platform in folder: /Users/hans/Library/Arduino15/packages/MegaCore/hardware/avr/3.0.0
Detecting libraries used...
/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega649 -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_ATmega649 -DARDUINO_ARCH_AVR -Wextra -flto -g -I/Users/hans/Library/Arduino15/packages/MegaCore/hardware/avr/3.0.0/cores/MCUdude_corefiles -I/Users/hans/Library/Arduino15/packages/MegaCore/hardware/avr/3.0.0/variants/64-pin-avr /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_425295/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega649 -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_ATmega649 -DARDUINO_ARCH_AVR -Wextra -flto -g -I/Users/hans/Library/Arduino15/packages/MegaCore/hardware/avr/3.0.0/cores/MCUdude_corefiles -I/Users/hans/Library/Arduino15/packages/MegaCore/hardware/avr/3.0.0/variants/64-pin-avr /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_425295/sketch/Blink.ino.cpp -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_425295/preproc/ctags_target_for_gcc_minus_e.cpp
[... cut ...]
         with 952 bytes in 1 section within [0, 0x3b7]
         using 4 pages and 72 pad bytes
avrdude: writing 952 bytes flash ...
Writing | ################################################## | 100% 0.31s
avrdude: 952 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_425295/Blink.ino.hex
Reading | ################################################## | 100% 0.16s
avrdude: 952 bytes of flash verified

avrdude done.  Thank you.

```

closed 2024-11-20 by MCUdude