MCUdude/MegaCore #209: Original Mega 2560 bootloader

aikopras, 2023-12-27:
Hi Hans

I moved to a new machine and installed today the latest Megacore release (V3). I noticed that I was no longer able to upload sketches on original Arduino Mega boards (via the onboard USB connector and its associated Atmega08). 
After comparison with my old machine (Megacore V2.2.0), I noticed that the old bootloader option "Original Mega 2560 bootloader" no longer exists. I compared both boards.txt files, and could confirm that this option indeed disappeared.

My question is whether this option could come back? If not, what would be the best way to proceed? is the only option to put Urboot on the original Arduino Mega board?

Maybe a little of context. I wrote some software, that others (with limited Arduino knowledge) would like to use. For them clear instructions on how to install my software via the standard USB port are needed. Asking them to buy a programmer (like USBasp) and flash a new bootloader on the 2560 processor via the 6 pin ICSP connector may not work. The past possibility to use the bootloader that already exists on the Arduino Mega board was quite convenient.

Thanks, Aiko

MCUdude, 2023-12-27:
Hi!

It's correct that I removed the legacy stk500v2 bootloader option from the v3.0.0 version of MegaCore. I did it to simplify the tools menu, and also to push users towards using the new but superior Urboot bootloader.

You can always ask them to install an older version of MegaCore or create a script that calls Avrdude directly. You can also use PlatformIO, which allows you to specify whatever programmer or bootloader you want.

The main difference between v3.0.0 and v2.2.3 versions is the migration to Urboot. If you downgrade to v2.2.3, you don't miss out on any core features apart from the new bootloader.

However, what I _could_ do for the future v3.0.1 release is to provide a legacy upload option for the default Arduino MEGA bootloader. I would recommend absolutely everyone to replace their default bootloader, but if this isn't an option, MegaCore could still be used. I'll probably implement under the _Programmers_ menu.

aikopras, 2023-12-28:
Thanks!
I downgraded to v2.2.3 and is now working again. :-)

And yes, I would very much like to see an option in v3.0.1 to "somehow" be able to use the old (original) Arduino Mega boot loader. For users of my software that have little experience with software and Arduino's, it would be nice if they are able to use the Arduino Mega board out of the box, thus without changing the bootloader. For them the Arduino IDE may be familiar, but they may not know PlatformIO. 

 Bye, Aiko

closed 2023-12-28 by aikopras

MCUdude, 2023-12-30:
I've just released v3.0.1
In this version, it's possible to upload using the original Arduino MEGA bootloader by selecting "Legacy Arduino MEGA bootloader" programmer option and clicking "Upload using programmer" to upload code using the default bootloader. However, I'll highly recommend replacing it with Urboot if possible.