MCUdude/MegaCore #211: Doesn't work on arduino mega 2560 (clone)

belpyro, 2024-01-04:
I tried several megas, and none worked after the bootloader replacement. MiniCore works properly with UNO (cloned).
The error looks like:
```
Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATmega2560
avrdude: AVR device initialized and ready to accept instructions
avrdude ur_initstruct() error: unknown bootloader ... please specify -xbootsize=<num>

avrdude avr_signature() error: unable to read signature data for part ATmega2560, rc=-1
avrdude main() error: unable to read signature data, rc=-1
```
My settings are:
[image]

Could someone route me on the right way? Thx!
P.s. I need clarification that the bootloader is always the same. Fuses changed only. Is it correct?

MCUdude, 2024-01-04:
... And you have burned the bootloader bundled with Megacore, using your USBasp programmer?

belpyro, 2024-01-06:
> ... And you have burned the bootloader bundled with Megacore, using your USBasp programmer?

Yes. That's correct. As I mentioned before, the same approach for UNO works fine. Thx!