MCUdude/MegaCore #216: ATmega64a serial upload problem

ITstreet1, 2024-06-01:
Hi,

I need help to make ATmega64a work. I made a board with the ATmega64a, using a design reference from here for basic electronics components. 

ATmega64a without a bootloader

The bootloader is programmed successfully by an Arduino Uno as an ISP. I can also upload any sketch using the Arduino Uno as a programmer this way, and some basic examples I tried worked just fine.

ATmega64a with bootloader
With the bootloader on board, I can not upload any example over the serial communication (I use CH340 on this board). I get:

> avrdude warning: attempt 1 of 10: not in sync
> avrdude warning: attempt 2 of 10: not in sync
> avrdude warning: attempt 3 of 10: not in sync
> avrdude warning: attempt 4 of 10: not in sync
> avrdude warning: attempt 5 of 10: not in sync
> avrdude warning: attempt 6 of 10: not in sync
> avrdude warning: attempt 7 of 10: not in sync
> avrdude warning: attempt 8 of 10: not in sync
> avrdude warning: attempt 9 of 10: not in sync
> avrdude warning: attempt 10 of 10: not in sync
> avrdude warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
> avrdude error: unable to open programmer urclock on port COM19
> avrdude error: unable to open programmer urclock on port COM19

Now, this suggests I don't have good communication between MCU and UART, but, when I upload the AnalogReadSerial example by an Arduino Uno as an ISP, I can read the data over the Serial terminal. So, UART looks ok.

Some light here, please.
I can provide any additional info if needed.

MCUdude, 2024-06-01:
What does your auto reset circuitry looks like? Do you have a schematic for it?

ITstreet1, 2024-06-01:
Here is the schematic:
[image]

I just checked the soldered joints. Everything is fine. I thought the DTR line was not ok, but it looks like it is.

MCUdude, 2024-06-01:
Can you try 250000 baud as the upload speed in the Arduino IDE settings?

ITstreet1, 2024-06-02:
Hi,

I have tried all baud rates, but the error message is the same.

MCUdude, 2024-06-02:
Can you try downgrading to version v2.2.3 and re-burn the bootloader? If this works, then it's related to the bootloader used in v3, Urboot.

ITstreet1, 2024-06-02:
Hi,

Ok just did it. Now, it behaves differently after a new bootloader. The LED on D13 is fading. With the ver3, it didn't.

But, the problem remains:

> avrdude error: programmer is not responding
> avrdude warning: attempt 1 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 2 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 3 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 4 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 5 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 6 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 7 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 8 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 9 of 10: not in sync: resp=0x00
> avrdude error: programmer is not responding
> avrdude warning: attempt 10 of 10: not in sync: resp=0x00
> avrdude error: unable to open programmer arduino on port COM19
> avrdude error: unable to open programmer arduino on port COM19

MCUdude, 2024-06-02:
The bahaviour you're seeing is correct. But what happens if you reduce the value of R4 to 100 ohms or so? I suspect 10k is way too high.

ITstreet1, 2024-06-02:
I get the value from here:
https://camo.githubusercontent.com/61f731fbae63fc0c4452657f1061d97b7a5e4a3b3b780b01535c5328afddb50c/68747470733a2f2f692e696d6775722e636f6d2f387034684e396e2e706e67

I will try and let you know

ITstreet1, 2024-06-02:
OK,
I changed it to 100R and it worked just fine. I also updated the boards to a ver3, and it works also. I will do some additional tests, but I don't see reasons not to work.

Please fix the minimal setup diagram.

ITstreet1, 2024-06-02:
Ok, after additional tests, everything looks just fine. The problem was in a wrong resistor value. Please fix it on a diagram.

closed 2024-06-02 by ITstreet1

tomaskovacik, 2024-06-02:
if internal pull up is set(RS232 on and idle)  with pull up resistor is 10k and 10k protection resistor then if CH340 pull his tx pin low you get nice 2.5V on pin, which is  probably still logic ONE. In case of 50k pull ups it is nice 1V which for region of no mans land, only if 100k internal pull ups are used then you can used 10k protection resistors ... so do your calculations ... also in case of 100Ohm resistor and 5V logic you are sinking 50mA into that pin, so check datasheet if it is ok, probably on edge ... 

anyway 1k is way to go, or use diode, serial is by design only working by pulling RX pins down  (idle voltage is VCC)

MCUdude, 2024-06-02:
The problem isn't really the 10k resistor, but the fact that you're trying to drive the LED though that resistor. That won't work for obvious reasons, and the signal the microcontroller is receiving is all messed up.

The reason why I added that resistor in the first place is because the RXD0/TXD0 pins are shared with the ISP programmer, and some ISP programmers will complain if the USB to serial chip is "loading" the data lines.

tomaskovacik, 2024-06-02:
oh I see he put RX LED(3) on wrong side of R4, should be connected to pin 2 of CH340, I did not spot that!