MCUdude/MegaCore #220: XMEM interface support

r4wabr, 2024-12-30:
I came across the fact that in the platformio environment it is not possible to fully program the AVR of the Mega family with the XMEM interface. I would like to have such an opportunity.

MCUdude, 2024-12-30:
I've never used this interface, but it's using avr-gcc underneath, where the user has access to the compiler flags. So if you know how to set up XMEM in, say, a makefile, then it may be possible to port this over to PlatformIO.

r4wabr, 2025-01-02:
compiler.ldflags=-Wl,--defsym=__heap_start=0x808000,--defsym=__heap_end=0x80FFFF

MCUdude, 2025-01-02:
With PlatformIO it should be able to do:

```ini
; Platformio.ini

build_flags = 
  -Wl,--defsym=__heap_start=0x808000,--defsym=__heap_end=0x80FFFF
```

If that's not working, it looks like you can use an [extra_script](https://docs.platformio.org/en/latest/scripting/examples/extra_linker_flags.html).