MCUdude/MegaCore #224: Warning: attempt 1 of 10: not in sync atmega64 atmega128

Daniel-hnr, 2025-03-31:
Hello everybody, I have a lot of ATmega128 and ATmega64 microcontrollers. I successfully burned the bootloader on them, but after uploading code once, I can’t upload anymore. I have added everything in the schematic.
Using port            : COM3
Using programmer      : urclock
Setting baud rate     : 115200
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port COM3 for programmer urclock

Avrdude done.  Thank you.
Failed uploading: uploading error: exit status 1

MCUdude, 2025-03-31:
Is your auto-reset circuitry correct (see minimal setup in the README)? Do you have a schematic I can see, and/or perhaps a picture of the board?

Daniel-hnr, 2025-03-31:
In board i don't add things in minimal setup but I add them between USB to serial and PCB and I get the error again

[image]

MCUdude, 2025-03-31:
I can't see any auto reset circuity. That's a 100nF capacitor between the DTR pin on your USB to serial adapter and a protection diode. It won't work without it. Manually timing the reset pulse is very hard.

Daniel-hnr, 2025-04-01:
I added a 100nF capacitor, but I'm still getting an error.

closed 2025-04-08 by Daniel-hnr